lgs2007m/Actions-OpenWrt #11: JDC-AX1800 Pro亚瑟刷U-boot后,无法进入web页面

weah2000, 2024-05-31:
大神，你好
我5月底新购的JDC-AX1800 Pro亚瑟64G，原厂固件r2242，先按照别的教程开启SSH后，再按照本仓库的刷机教程，刷了uboot-JDC_AX1800_Pro-AX6600_Athena-20240510这个uboot后，并按照教程刷了gpt-JDC_AX1800_Pro_dual-boot_rootfs512M_no-last-partition双分区，校验都没问题，重启后，灯是正常的闪5红变蓝，电脑的IP设置为192.168.1.2，子网掩码255.255.255.0，但无法进入192.168.1.1这个地址，ping也不通。这种情况是不是变转了？有没有办法救砖？我试图用USB 9008的方法不拆机救砖，但电脑找不到设备。

weah2000, 2024-05-31:
分区做了备份，如果没法进入u-boot刷固件，要如何恢复分区?

xiake8, 2024-05-31:
进uboot的时候 按住重置键  再通电  善绿灯才是进了

weah2000, 2024-05-31:
> 进uboot的时候 按住重置键 再通电 善绿灯才是进了

`电脑网线接路由器lan口，电脑ip设置192.168.1.2/24，然后路由器上电，按住reset按键不放，LED灯红色会闪烁5次，然后LED灯会变蓝色。
LED变蓝色说明uboot failsafe webui已经开启，浏览器打开登录地址192.168.1.1`

教程上的描述是变蓝色就是webui开启了，我也是到了这步，但192.168.1.1却怎么也访问不了

lgs2007m, 2024-06-01:
网线连接路由器lan口（黑色的网口），电脑ip设置192.168.1.2，掩码255.255.255.0，网关不需要，点击确定然后点击关闭，关闭以太网设置窗口，这样设置才能生效。设置好后可以查看以太网详细信息，看看ip是否设置上去了。接着路由器上电，按reset不放等待蓝灯亮起，再松开reset即可。不行就断电再重新上电尝试。

lgs2007m, 2024-06-01:
> 进uboot的时候 按住重置键 再通电 善绿灯才是进了

我改的uboot是亮蓝灯表示uboot webui已开启，其他人的不懂，看其他教程的说明

weah2000, 2024-06-01:
> 网线连接路由器lan口（黑色的网口），电脑ip设置192.168.1.2，掩码255.255.255.0，网关不需要，点击确定然后点击关闭，关闭以太网设置窗口，这样设置才能生效。设置好后可以查看以太网详细信息，看看ip是否设置上去了。接着路由器上电，按reset不放等待蓝灯亮起，再松开reset即可。不行就断电再重新上电尝试，还不行，检查的网口、网线。

已确认ipconfig信息，网线也换过了，试了好几次，依旧无效
故障情况可见视频，https://b23.tv/RqrSalN

lgs2007m, 2024-06-01:
> > 网线连接路由器lan口（黑色的网口），电脑ip设置192.168.1.2，掩码255.255.255.0，网关不需要，点击确定然后点击关闭，关闭以太网设置窗口，这样设置才能生效。设置好后可以查看以太网详细信息，看看ip是否设置上去了。接着路由器上电，按reset不放等待蓝灯亮起，再松开reset即可。不行就断电再重新上电尝试，还不行，检查的网口、网线。
> 
> 已确认ipconfig信息，网线也换过了，试了好几次，依旧无效 故障情况可见视频: b站: https://b23.tv/RqrSalN youtube: https://youtu.be/nAGRjOXaQ78

看了，没发现什么问题……
上一版uboot我测试了雅典娜、亚瑟都没有问题。
这版uboot我只改了刷机部分的一句命令，按理说不影响进入webui。
因为现在我只有雅典娜，测试不了亚瑟，我问了下群友，他们反馈亚瑟这版20240510 uboot正常使用。
实在不行，你只能拆机接TTL看输出什么，然后再看看怎么解决了。

weah2000, 2024-06-02:
> > > 网线连接路由器lan口（黑色的网口），电脑ip设置192.168.1.2，掩码255.255.255.0，网关不需要，点击确定然后点击关闭，关闭以太网设置窗口，这样设置才能生效。设置好后可以查看以太网详细信息，看看ip是否设置上去了。接着路由器上电，按reset不放等待蓝灯亮起，再松开reset即可。不行就断电再重新上电尝试，还不行，检查的网口、网线。
> > 
> > 
> > 已确认ipconfig信息，网线也换过了，试了好几次，依旧无效 故障情况可见视频: b站: https://b23.tv/RqrSalN youtube: https://youtu.be/nAGRjOXaQ78
> 
> 看了，没发现什么问题…… 上一版uboot我测试了雅典娜、亚瑟都没有问题。 这版uboot我只改了刷机部分的一句命令，按理说不影响进入webui。 因为现在我只有雅典娜，测试不了亚瑟，我问了下群友，他们反馈亚瑟这版20240510 uboot正常使用。 实在不行，你只能拆机接TTL看输出什么，然后再看看怎么解决了。 如果你可以进入系统，可以换个旧版的uboot试试。

好的，我先找个ttl转接板

ldd200888, 2024-06-02:
相同的问题，也是死活进不了webui，ping都不通。。。

lgs2007m, 2024-06-02:
> 相同的问题，也是死活进不了webui，ping都不通。。。

看到留言，我马上找了一位群友测试下亚瑟刷20240510的uboot，可以正常进uboot webui，不知道你们到底什么情况了

ldd200888, 2024-06-02:
> > 相同的问题，也是死活进不了webui，ping都不通。。。
> 
> 看到留言，我马上找了一位群友测试下亚瑟刷20240510的uboot，可以正常进uboot webui，不知道你们到底什么情况了

请问老版的uboot哪儿找？

lgs2007m, 2024-06-02:
> > > 相同的问题，也是死活进不了webui，ping都不通。。。
> > 
> > 
> > 看到留言，我马上找了一位群友测试下亚瑟刷20240510的uboot，可以正常进uboot webui，不知道你们到底什么情况了
> 
> 请问老版的uboot哪儿找？
[https://github.com/lgs2007m/Actions-OpenWrt/tree/main/刷机教程](https://github.com/lgs2007m/Actions-OpenWrt/tree/main/刷机教程)

u2shana, 2024-06-02:
想问一下ImmortalWrt的uboot单独刷就行了么，需不要需要配套他的bl2和分区表

lgs2007m, 2024-06-02:
> 想问一下ImmortalWrt的uboot单独刷就行了么，需不要需要配套他的bl2和分区表

不知道你说哪个机子……有bl2的是百里吧？ImmortalWrt主线的uboot和bl2、gpt应该是要配套的。具体去ImmortalWrt仓库看吧。
还有ImmortalWrt的天灵大佬单独做的带DHCP的uboot是使用hanwckf大佬的源码做的，是使用闭源固件的，和主线的不一样。我改的uboot也是hanwckf大佬仓库做的。

u2shana, 2024-06-02:
> > 想问一下ImmortalWrt的uboot单独刷就行了么，需不要需要配套他的bl2和分区表
> 
> 不知道你说哪个机子……有bl2的是百里吧？ImmortalWrt主线的uboot和bl2、gpt应该是要配套的。具体去ImmortalWrt仓库看吧。 还有ImmortalWrt的天灵大佬单独做的带DHCP的uboot是使用hanwckf大佬的源码做的，是使用闭源固件的，和主线的不一样。我改的uboot也是hanwckf大佬仓库做的。

就是说天灵大佬的uboot，刷成他的uboot，bl2和分区表要不要重刷成他提供的

lgs2007m, 2024-06-02:
> > > 想问一下ImmortalWrt的uboot单独刷就行了么，需不要需要配套他的bl2和分区表
> > 
> > 
> > 不知道你说哪个机子……有bl2的是百里吧？ImmortalWrt主线的uboot和bl2、gpt应该是要配套的。具体去ImmortalWrt仓库看吧。 还有ImmortalWrt的天灵大佬单独做的带DHCP的uboot是使用hanwckf大佬的源码做的，是使用闭源固件的，和主线的不一样。我改的uboot也是hanwckf大佬仓库做的。
> 
> 就是说天灵大佬的uboot，刷成他的uboot，bl2和分区表要不要重刷成他提供的

不用

ldd200888, 2024-06-03:
> > > > 相同的问题，也是死活进不了webui，ping都不通。。。
> > > 
> > > 
> > > 看到留言，我马上找了一位群友测试下亚瑟刷20240510的uboot，可以正常进uboot webui，不知道你们到底什么情况了
> > 
> > 
> > 请问老版的uboot哪儿找？
> > [https://github.com/lgs2007m/Actions-OpenWrt/tree/main/刷机教程](https://github.com/lgs2007m/Actions-OpenWrt/tree/main/%E5%88%B7%E6%9C%BA%E6%95%99%E7%A8%8B)

这个目录我没找见老版的uboot，只有0510的这个版本。 另外补充下一些信息：

1. 我的路由器是刚到货的全新亚瑟，原始系统早于r2181，好像是r20xx版本。
2. 在刷uboot前，未联网，仅修改了wifi名字和密码
3. 所有操作均严格按照文档说明操作
4. 网线无问题，在正常启动的时候，网线连接可以正常访问路由器管理页面192.168.68.1
5. 刷了uboot后，路由器还可以正常进入正常系统。
6. 刷了gpt分区表后，无法进入正常系统（无法修改密码和wifi名字）

猜想：

1. 这个问题是不是和初始系统版本有关？这个版本很老，可能都是21年的

请问群是哪个？我也想进去。。

ldd200888, 2024-06-03:
> 好的，我先找个ttl转接板

咋样？ttl有啥错误？

lgs2007m, 2024-06-03:
> 这个目录我没找见老版的uboot，只有0510的这个版本。 另外补充下一些信息：
>
> 1. 我的路由器是刚到货的全新亚瑟，原始系统早于r2181，好像是r20xx版本。
> 2. 在刷uboot前，未联网，仅修改了wifi名字和密码
> 3. 所有操作均严格按照文档说明操作
> 4. 网线无问题，在正常启动的时候，网线连接可以正常访问路由器管理页面192.168.68.1
> 5. [...]

我搞错了，你要找老版，我看成老板了……
旧版在这里 https://github.com/lgs2007m/Actions-OpenWrt/tree/09fe89489657906d1e5b687d03e59244fcdae69a
刷gpt后一般需要重新刷固件，因为分区改变，系统数据可能已经不对了，和系统版本无关。
MT798X交流Q群 837365344

ldd200888, 2024-06-03:
我再刷下老板uboot，看咋样。

weah2000, 2024-06-03:
> 咋样？ttl有啥错误？

看了拆机教程，我放弃拆机了，新买的还可以换货，已经申请售后换机了

lgs2007m, 2024-06-03:
> 看了拆机教程，我放弃拆机了，新买的还可以换货，已经申请售后换机了

嗯嗯

ldd200888, 2024-06-04:
> 我再刷下老板uboot，看咋样。

老版的uboot一样，也是进不去webui

lgs2007m, 2024-06-04:
> > 我再刷下老板uboot，看咋样。
> 
> 老版的uboot一样，也是进不去webui

老版我自己测试过没问题的。之前有个群友说比较难进webui，我立马试了下，一下就进去了。
你能拆机的话，可以看看TTL输出什么，不行就去恩山找更旧版的uboot或者大雕的uboot试试

ldd200888, 2024-06-04:
请问只刷大雕的uboot后，不刷gpt分区表，还能进原始系统吗？

lgs2007m, 2024-06-04:
> 请问只刷大雕的uboot后，不刷gpt分区表，还能进原始系统吗？

没试过。理论上大家都是使用gl开源的ipq60xx uboot源码制作的，都是可以的

ldd200888, 2024-06-04:
谢谢~~~

allonmymind, 2024-06-05:
我也同样遇到问题，第一次刷您的uboot可以正常进web页面。一天后就进不去了，重刷也不行，但是刷旧的u-boot.mbn那个文件就可以正常打开，最开始我以为是我机器的问题，看到这个issue才知道原来不止我一个遇到了

lgs2007m, 2024-06-05:
> 我也同样遇到问题，第一次刷您的uboot可以正常进web页面。一天后就进不去了，重刷也不行，但是刷旧的u-boot.mbn那个文件就可以正常打开，最开始我以为是我机器的问题，看到这个issue才知道原来不止我一个遇到了

这不可能吧，刷进入uboot数据是不会变的，不会第一次可以隔一天不行。你刷uboot后还是使用原厂系统跑分吗？

ldd200888, 2024-06-05:
> > > 我再刷下老板uboot，看咋样。
> > 
> > 
> > 老版的uboot一样，也是进不去webui
> 
> 老版我自己测试过没问题的。之前有个群友说比较难进webui，我立马试了下，一下就进去了。 你能拆机的话，可以看看TTL输出什么，不行就去恩山找更旧版的uboot或者大雕的uboot试试

1. 我刷了大雕的uboot，是可以进webui的
2. 然后再直接刷新版uboot，还是进不去
3. 用大雕的uboot里面刷新版uboot，还是进不去。。

最后就刷了大雕的uboot。。

allonmymind, 2024-06-05:
> > 我也同样遇到问题，第一次刷您的uboot可以正常进web页面。一天后就进不去了，重刷也不行，但是刷旧的u-boot.mbn那个文件就可以正常打开，最开始我以为是我机器的问题，看到这个issue才知道原来不止我一个遇到了
> 
> 这不可能吧，刷进入uboot数据是不会变的，不会第一次可以隔一天不行。你刷uboot后还是使用原厂系统跑分吗？

我是istore系统刷用ssh方法刷的您的uboot，应该是刷完以后改了大分区，不知道和这个有没有关系，现在进系统后日志会有这个报错：
root@iStoreOS:~# dmesg | grep error
[    2.144729] UBIFS error (pid: 1): cannot open "ubi0:rootfs", error -19
[   22.964140] Subsystem error monitoring/handling services are up
[   32.014082] Subsystem error monitoring/handling services are up
[  151.166555] Subsystem error monitoring/handling services are up
[  395.579422] Subsystem error monitoring/handling services are up
[ 1140.617515] Subsystem error monitoring/handling services are up
root@iStoreOS:~#

allonmymind, 2024-06-05:
> > > > 我再刷下老板uboot，看咋样。
> > > 
> > > 
> > > 老版的uboot一样，也是进不去webui
> > 
> > 
> > 老版我自己测试过没问题的。之前有个群友说比较难进webui，我立马试了下，一下就进去了。 你能拆机的话，可以看看TTL输出什么，不行就去恩山找更旧版的uboot或者大雕的uboot试试
> 
> 1. 我刷了大雕的uboot，是可以进webui的
> 2. 然后再直接刷新版uboot，还是进不去
> 3. 用大雕的uboot里面刷新版uboot，还是进不去。。
> 
> 最后就刷了大雕的uboot。。

大雕的uboot是640K还是630K？

lgs2007m, 2024-06-05:
> 我是istore系统刷用ssh方法刷的您的uboot，应该是刷完以后改了大分区，不知道和这个有没有关系，现在进系统后日志会有这个报错： root@iStoreOS:~# dmesg | grep error [ 2.144729] UBIFS error (pid: 1): cannot open "ubi0:rootfs", error -19 [ 22.964140] Subsystem [...]

这些报错和uboot没啥关系，uboot是引导启动系统的。
我让群友测试，未发现亚瑟uboot进不了webui……不知道你们的是什么情况了。
我只有雅典娜，uboot正常使用。之前测试亚瑟uboot上一版，没什么问题，亚瑟已经出掉了……
我想去JD买个亚瑟重新刷uboot看看了。

allonmymind, 2024-06-05:
> 这些报错和uboot没啥关系，uboot是引导启动系统的。 我让群友测试，未发现亚瑟uboot进不了webui……不知道你们的是什么情况了。 我只有雅典娜，uboot正常使用。之前测试亚瑟uboot上一版，没什么问题，亚瑟已经出掉了…… 我想去JD买个亚瑟重新刷uboot看看了。

感觉像是分区问题导致的，我的第一条报错也许和进不了uboot都是分区问题引起的

lgs2007m, 2024-06-05:
> 感觉像是分区问题导致的，我的第一条报错也许和进不了uboot都是分区问题引起的

用的是我教程中的分区表吗？提供下你用的分区表看看
不过感觉不应该，因为如果分区表有问题，系统是启动不了的

allonmymind, 2024-06-05:
> 用的是我教程中的分区表吗？提供下你用的分区表看看 不过感觉不应该，因为如果分区表有问题，系统是启动不了的

是的，用的亚瑟2G那个
root@iStoreOS:~# blkid
/dev/loop0: LABEL="rootfs_data" UUID="268e58b7-b101-4942-8fd0-4f32c07deb34" BLOCK_SIZE="4096" TYPE="ext4"
/dev/mmcblk0p1: PARTLABEL="0:SBL1" PARTUUID="23691061-ce82-a7f9-e63d-92b9cdf584f6"
/dev/mmcblk0p2: PARTLABEL="0:BOOTCONFIG" PARTUUID="89c1df88-fb18-5142-477d-de6878281347"
/dev/mmcblk0p3: PARTLABEL="0:BOOTCONFIG1" PARTUUID="0802fa1e-92bf-ed7c-f7e0-20803765efd3"
/dev/mmcblk0p4: PARTLABEL="0:QSEE" PARTUUID="0a81139c-3bb6-8db2-fae8-aaba889f4861"
/dev/mmcblk0p5: PARTLABEL="0:QSEE_1" PARTUUID="4ba8c830-46ab-cf60-d7fc-2301436c266b"
/dev/mmcblk0p6: PARTLABEL="0:DEVCFG" PARTUUID="6402e55f-c9f4-438b-7c1c-2bba6467f752"
/dev/mmcblk0p7: PARTLABEL="0:DEVCFG_1" PARTUUID="a66ed3dd-0348-b9e1-58ab-2d10459cff71"
/dev/mmcblk0p8: PARTLABEL="0:RPM" PARTUUID="084f94d8-8243-d676-2bcf-0b088dd6f58e"
/dev/mmcblk0p9: PARTLABEL="0:RPM_1" PARTUUID="7296dd7d-817e-d7ac-b818-ab3100620029"
/dev/mmcblk0p10: PARTLABEL="0:CDT" PARTUUID="60ba117e-3af5-3d6d-f41b-f90f8af17982"
/dev/mmcblk0p11: PARTLABEL="0:CDT_1" PARTUUID="0b4dc452-ec49-8825-1b12-344036039c9e"
/dev/mmcblk0p12: PARTLABEL="0:APPSBLENV" PARTUUID="65d77f3a-9863-97d7-0c00-a1826aa337ef"
/dev/mmcblk0p13: PARTLABEL="0:APPSBL" PARTUUID="a656941f-4a5a-8e97-b038-3c56ca29cb26"
/dev/mmcblk0p14: PARTLABEL="0:APPSBL_1" PARTUUID="ce2236d3-f9c8-9e3b-22d9-a0c67a008ea4"
/dev/mmcblk0p15: PARTLABEL="0:ART" PARTUUID="ae4e3c02-5a5b-8f0d-f829-db226febe307"
/dev/mmcblk0p16: PARTLABEL="0:HLOS" PARTUUID="1d04de0e-5189-61b6-57fb-b16554d1cdf3"
/dev/mmcblk0p17: PARTLABEL="0:HLOS_1" PARTUUID="6c478b91-3026-043e-67b2-6bd87f42642f"
/dev/mmcblk0p18: TYPE="squashfs" PARTLABEL="rootfs" PARTUUID="49ce1048-1499-ae66-4240-f15d706a2598"
/dev/mmcblk0p19: PARTLABEL="0:WIFIFW" PARTUUID="4609454a-2ea6-cfa6-c005-531838befaf6"
/dev/mmcblk0p20: PARTLABEL="rootfs_1" PARTUUID="3db96119-20c1-90b8-3dd6-43f241ac87b8"
/dev/mmcblk0p21: PARTLABEL="0:WIFIFW_1" PARTUUID="59e7b7d5-03c4-caba-d2e8-3661ff6ab7bd"
/dev/mmcblk0p22: PARTLABEL="rootfs_data" PARTUUID="87d7a95d-007b-f9d0-3c54-dd4f9140d35e"
/dev/mmcblk0p23: PARTLABEL="0:ETHPHYFW" PARTUUID="c03e0f0b-3f28-ec31-0c36-d119817baeb0"
/dev/mmcblk0p24: PARTLABEL="plugin" PARTUUID="260f092b-3b3a-3710-4f48-846ec245d3a1"
/dev/mmcblk0p25: PARTLABEL="log" PARTUUID="837fae9b-8118-b2ec-f195-877a5a0133cd"
/dev/mmcblk0p26: PARTLABEL="swap" PARTUUID="7f63406e-c870-c8b0-489a-45cdec955611"
/dev/mmcblk0p27: UUID="e9f9ad5b-8bda-4c1a-a9eb-1bbad037044c" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="cc852b77-5469-4548-9c36-9e3ec24afd0e"
root@iStoreOS:~# fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 57.12 GiB, 61329113088 bytes, 119783424 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 98101B32-BBE2-4BF2-A06E-2BB33D000C20

Device            Start       End   Sectors  Size Type
/dev/mmcblk0p1       34      1569      1536  768K unknown
/dev/mmcblk0p2     1570      2081       512  256K unknown
/dev/mmcblk0p3     2082      2593       512  256K unknown
/dev/mmcblk0p4     2594      6177      3584  1.8M unknown
/dev/mmcblk0p5     6178      9761      3584  1.8M unknown
/dev/mmcblk0p6     9762     10273       512  256K unknown
/dev/mmcblk0p7    10274     10785       512  256K unknown
/dev/mmcblk0p8    10786     11297       512  256K unknown
/dev/mmcblk0p9    11298     11809       512  256K unknown
/dev/mmcblk0p10   11810     12321       512  256K unknown
/dev/mmcblk0p11   12322     12833       512  256K unknown
/dev/mmcblk0p12   12834     13345       512  256K unknown
/dev/mmcblk0p13   13346     14625      1280  640K unknown
/dev/mmcblk0p14   14626     15905      1280  640K unknown
/dev/mmcblk0p15   15906     16417       512  256K unknown
/dev/mmcblk0p16   16418     28705     12288    6M unknown
/dev/mmcblk0p17   28706     40993     12288    6M unknown
/dev/mmcblk0p18   40994   4235297   4194304    2G unknown
/dev/mmcblk0p19 4235298   4243489      8192    4M unknown
/dev/mmcblk0p20 4243490   4366369    122880   60M unknown
/dev/mmcblk0p21 4366370   4374561      8192    4M unknown
/dev/mmcblk0p22 4374562   4415521     40960   20M unknown
/dev/mmcblk0p23 4415522   4416545      1024  512K unknown
/dev/mmcblk0p24 4416546   4595745    179200 87.5M unknown
/dev/mmcblk0p25 4595746   5210145    614400  300M unknown
/dev/mmcblk0p26 5210146   6258721   1048576  512M unknown
/dev/mmcblk0p27 6260736 119783390 113522655 54.1G Linux filesystem
root@iStoreOS:~#

lgs2007m, 2024-06-05:
> 是的，用的亚瑟2G那个 root@iStoreOS:~# blkid /dev/loop0: LABEL="rootfs_data" UUID="268e58b7-b101-4942-8fd0-4f32c07deb34" BLOCK_SIZE="4096" TYPE="ext4" /dev/mmcblk0p1: PARTLABEL="0:SBL1" [...]

那没啥问题呀，我都检查过分区的大小

allonmymind, 2024-06-05:
> 那没啥问题呀，我都检查过分区的大小

刚开始我还怀疑是emmc问题，后来换了个emmc也是进不了您的uboot

lgs2007m, 2024-06-05:
> 刚开始我还怀疑是emmc问题，后来换了个emmc也是进不了您的uboot

囧，一部分人说没问题，可以进，一部分人说进不了……
突然之间少了些自信，哈哈。等我后面买个机子再测试看看吧

allonmymind, 2024-06-05:
> 囧，一部分人说没问题，可以进，一部分人说进不了…… 突然之间少了些自信，哈哈。等我后面买个机子再测试看看吧

有没有可能是文件大小问题，那个普通uboot只有630k，能进。这个640k的进不了，连192.168.1.1都ping不通。

lgs2007m, 2024-06-05:
> 有没有可能是文件大小问题，那个普通uboot只有630k，能进。这个640k的进不了，连192.168.1.1都ping不通。

不会，分区大小就是640KB，630KB是编译出来的原始文件，我是直接将原始文件尾部填充0扩大到640KB了。这个文件可以看做分区的备份。
这样做之后，刷uboot相当于先擦除分区，再写入uboot原始文件，dd刷写不清除所有扇区，这样可以避免分区尾部有残留的数据不会被覆盖（有残留这个其实不影响，我只是有数据洁癖）。

allonmymind, 2024-06-05:
> 不会，分区大小就是640KB，630KB是编译出来的原始文件，我是直接将原始文件尾部填充0扩大到640KB了。这个文件可以看做分区的备份。 这样做之后，刷uboot相当于先擦除分区，再写入uboot原始文件，dd刷写不清除所有扇区，这样可以避免分区尾部有残留的数据不会被覆盖（有残留这个其实不影响，我只是有数据洁癖）。

ipq6000另一机型兆能M2能用这个uboot吗，我换个机器测试下看看

ldd200888, 2024-06-05:
大雕的uboot比你的功能少多了，不支持uboot中刷gpt，还不支持好几个呢，希望这个uboot兼容性再好一点，就完美了

lgs2007m, 2024-06-05:
> ipq6000另一机型兆能M2能用这个uboot吗，我换个机器测试下看看

不支持，只支持亚瑟、雅典娜

allonmymind, 2024-06-05:
> 不支持，只支持亚瑟、雅典娜

uboot的ttl信息如下

Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.XF.0.3-00100-IPQ60xxLZB-1
S - IMAGE_VARIANT_STRING=IPQ6018LA
S - OEM_IMAGE_VERSION_STRING=crm-ubuntu124
S - Boot Interface: eMMC
S - Secure Boot: Off
S - Boot Config @ 0x000a602c = 0x000002e3
S - JTAG ID @ 0x000a607c = 0x001390e1
S - OEM ID @ 0x000a6080 = 0x00000000
S - Serial Number @ 0x000a4128 = 0xa8409ac0
S - OEM Config Row 0 @ 0x000a4188 = 0x0000000000000000
S - OEM Config Row 1 @ 0x000a4190 = 0x0000000000000000
S - Feature Config Row 0 @ 0x000a4130 = 0x0000800018200021
S - Feature Config Row 1 @ 0x000a4138 = 0x02c3e83783000009
S - PBL Patch Ver: 1
S - I-cache: On
S - D-cache: On
B -      3413 - PBL, Start
B -       592 - bootable_media_detect_entry, Start
B -      4339 - bootable_media_detect_success, Start
B -     97393 - elf_loader_entry, Start
B -     97565 - auth_hash_seg_entry, Start
B -     98547 - auth_hash_seg_exit, Start
B -     99039 - elf_segs_hash_verify_entry, Start
B -    125058 - elf_segs_hash_verify_exit, Start
B -    129249 - auth_xbl_sec_hash_seg_entry, Start
B -    129393 - auth_xbl_sec_hash_seg_exit, Start
B -    135942 - xbl_sec_segs_hash_verify_entry, Start
B -    135942 - xbl_sec_segs_hash_verify_exit, Start
B -    136872 - PBL, End
B -    114100 - SBL1, Start
B -    256993 - GCC [RstStat:0x0, RstDbg:0x600000] WDog Stat : 0x4
B -    259433 - clock_init, Start
D -      3599 - clock_init, Delta
B -    268888 - boot_flash_init, Start
D -     54625 - boot_flash_init, Delta
B -    326197 - sbl1_ddr_set_default_params, Start
D -       244 - sbl1_ddr_set_default_params, Delta
B -    332511 - boot_config_data_table_init, Start
D -      2287 - boot_config_data_table_init, Delta - (575 Bytes)
B -    341630 - CDT Version:2,Platform ID:8,Major ID:3,Minor ID:2,Subtype:1
B -    347181 - Image Load, Start
D -      6619 - OEM_MISC Image Loaded, Delta - (0 Bytes)
B -    356514 - Image Load, Start
D -      5063 - PMIC Image Loaded, Delta - (0 Bytes)
B -    364414 - sbl1_ddr_set_params, Start
B -    369385 - CPR configuration: 0x555
B -    372588 - Pre_DDR_clock_init, Start
D -       213 - Pre_DDR_clock_init, Delta
D -         0 - sbl1_ddr_set_params, Delta
B -    407083 - Image Load, Start
D -       519 - APDP Image Loaded, Delta - (0 Bytes)
B -    425780 - Image Load, Start
D -       518 - QTI_MISC Image Loaded, Delta - (0 Bytes)
B -    428464 - Image Load, Start
D -       823 - Auth Metadata
D -       671 - Segments hash check
D -     11681 - QSEE Dev Config Image Loaded, Delta - (37154 Bytes)
B -    442006 - Image Load, Start
D -      6558 - Auth Metadata
D -     10370 - Segments hash check
D -     95038 - QSEE Image Loaded, Delta - (1439116 Bytes)
B -    537593 - Image Load, Start
D -       701 - Auth Metadata
D -       976 - Segments hash check
[... 120 lines cut ...]
eth0 PHY1 Down Speed :10 Half duplex
eth0 PHY2 Down Speed :10 Half duplex
eth0 PHY3 Down Speed :10 Half duplex
eth0 PHY4 Down Speed :10 Half duplex
eth0 PHY0 Down Speed :10 Half duplex
eth0 PHY1 Down Speed :10 Half duplex
eth0 PHY2 Down Speed :10 Half duplex
eth0 PHY3 Down Speed :10 Half duplex
eth0 PHY4 Down Speed :10 Half duplex
eth0 PHY0 Down Speed :10 Half duplex
eth0 PHY1 Down Speed :10 Half duplex
eth0 PHY2 Down Speed :10 Half duplex
eth0 PHY3 Down Speed :10 Half duplex
eth0 PHY4 Down Speed :10 Half duplex
eth0 PHY0 Down Speed :10 Half duplex
eth0 PHY1 Down Speed :10 Half duplex
eth0 PHY2 Down Speed :10 Half duplex
eth0 PHY3 up Speed :10 Full duplex
eth0 PHY4 Down Speed :10 Half duplex
ipq6018_eth_init: done
HTTP server is starting at IP: 192.168.1.1
done set host addr 0xa8c0 0x101
HTTP server is ready!

然后依然ping不通192.168.1.1

lgs2007m, 2024-06-05:
> > > > > > > > > > > > > > > 我也同样遇到问题，第一次刷您的uboot可以正常进web页面。一天后就进不去了，重刷也不行，但是刷旧的u-boot.mbn那个文件就可以正常打开，最开始我以为是我机器的问题，看到这个issue才知道原来不止我一个遇到了
> > > > > > > > > > > > > > 
> > > > > > > > > > > > > > 
> > > > > > > > > > > > > > 这不可能吧，刷进入uboot数据是不会变的，不会第一次可以隔一天不行。你刷uboot后还是使用原厂系统跑分吗？
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 我是istore系统刷用ssh方法刷的您的uboot，应该是刷完以后改了大分区，不知道和这个有没有关系，现在进系统后日志会有这个报错： root@iStoreOS:~# dmesg | grep error [ 2.144729] UBIFS error (pid: 1): cannot open "ubi0:rootfs", error -19 [ 22.964140] Subsystem error monitoring/handling services are up [ 32.014082] Subsystem error monitoring/handling services are up [ 151.166555] Subsystem error monitoring/handling services are up [ 395.579422] Subsystem error monitoring/handling services are up [ 1140.617515] Subsystem error monitoring/handling services are up root@iStoreOS:~#
> > > > > > > > > > > > 
> > > > > > > > > > > > 
> > > > > > > > > > > > 这些报错和uboot没啥关系，uboot是引导启动系统的。 我让群友测试，未发现亚瑟uboot进不了webui……不知道你们的是什么情况了。 我只有雅典娜，uboot正常使用。之前测试亚瑟uboot上一版，没什么问题，亚瑟已经出掉了…… 我想去JD买个亚瑟重新刷uboot看看了。
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > 感觉像是分区问题导致的，我的第一条报错也许和进不了uboot都是分区问题引起的
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > 用的是我教程中的分区表吗？提供下你用的分区表看看 不过感觉不应该，因为如果分区表有问题，系统是启动不了的
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > 是的，用的亚瑟2G那个 root@iStoreOS:~# blkid /dev/loop0: LABEL="rootfs_data" UUID="268e58b7-b101-4942-8fd0-4f32c07deb34" BLOCK_SIZE="4096" TYPE="ext4" /dev/mmcblk0p1: PARTLABEL="0:SBL1" PARTUUID="23691061-ce82-a7f9-e63d-92b9cdf584f6" /dev/mmcblk0p2: PARTLABEL="0:BOOTCONFIG" PARTUUID="89c1df88-fb18-5142-477d-de6878281347" /dev/mmcblk0p3: PARTLABEL="0:BOOTCONFIG1" PARTUUID="0802fa1e-92bf-ed7c-f7e0-20803765efd3" /dev/mmcblk0p4: PARTLABEL="0:QSEE" PARTUUID="0a81139c-3bb6-8db2-fae8-aaba889f4861" /dev/mmcblk0p5: PARTLABEL="0:QSEE_1" PARTUUID="4ba8c830-46ab-cf60-d7fc-2301436c266b" /dev/mmcblk0p6: PARTLABEL="0:DEVCFG" PARTUUID="6402e55f-c9f4-438b-7c1c-2bba6467f752" /dev/mmcblk0p7: PARTLABEL="0:DEVCFG_1" PARTUUID="a66ed3dd-0348-b9e1-58ab-2d10459cff71" /dev/mmcblk0p8: PARTLABEL="0:RPM" PARTUUID="084f94d8-8243-d676-2bcf-0b088dd6f58e" /dev/mmcblk0p9: PARTLABEL="0:RPM_1" PARTUUID="7296dd7d-817e-d7ac-b818-ab3100620029" /dev/mmcblk0p10: PARTLABEL="0:CDT" PARTUUID="60ba117e-3af5-3d6d-f41b-f90f8af17982" /dev/mmcblk0p11: PARTLABEL="0:CDT_1" PARTUUID="0b4dc452-ec49-8825-1b12-344036039c9e" /dev/mmcblk0p12: PARTLABEL="0:APPSBLENV" PARTUUID="65d77f3a-9863-97d7-0c00-a1826aa337ef" /dev/mmcblk0p13: PARTLABEL="0:APPSBL" PARTUUID="a656941f-4a5a-8e97-b038-3c56ca29cb26" /dev/mmcblk0p14: PARTLABEL="0:APPSBL_1" PARTUUID="ce2236d3-f9c8-9e3b-22d9-a0c67a008ea4" /dev/mmcblk0p15: PARTLABEL="0:ART" PARTUUID="ae4e3c02-5a5b-8f0d-f829-db226febe307" /dev/mmcblk0p16: PARTLABEL="0:HLOS" PARTUUID="1d04de0e-5189-61b6-57fb-b16554d1cdf3" /dev/mmcblk0p17: PARTLABEL="0:HLOS_1" PARTUUID="6c478b91-3026-043e-67b2-6bd87f42642f" /dev/mmcblk0p18: TYPE="squashfs" PARTLABEL="rootfs" PARTUUID="49ce1048-1499-ae66-4240-f15d706a2598" /dev/mmcblk0p19: PARTLABEL="0:WIFIFW" PARTUUID="4609454a-2ea6-cfa6-c005-531838befaf6" /dev/mmcblk0p20: PARTLABEL="rootfs_1" PARTUUID="3db96119-20c1-90b8-3dd6-43f241ac87b8" /dev/mmcblk0p21: PARTLABEL="0:WIFIFW_1" PARTUUID="59e7b7d5-03c4-caba-d2e8-3661ff6ab7bd" /dev/mmcblk0p22: PARTLABEL="rootfs_data" PARTUUID="87d7a95d-007b-f9d0-3c54-dd4f9140d35e" /dev/mmcblk0p23: PARTLABEL="0:ETHPHYFW" PARTUUID="c03e0f0b-3f28-ec31-0c36-d119817baeb0" /dev/mmcblk0p24: PARTLABEL="plugin" PARTUUID="260f092b-3b3a-3710-4f48-846ec245d3a1" /dev/mmcblk0p25: PARTLABEL="log" PARTUUID="837fae9b-8118-b2ec-f195-877a5a0133cd" /dev/mmcblk0p26: PARTLABEL="swap" PARTUUID="7f63406e-c870-c8b0-489a-45cdec955611" /dev/mmcblk0p27: UUID="e9f9ad5b-8bda-4c1a-a9eb-1bbad037044c" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="cc852b77-5469-4548-9c36-9e3ec24afd0e" root@iStoreOS:~# fdisk -l /dev/mmcblk0 Disk /dev/mmcblk0: 57.12 GiB, 61329113088 bytes, 119783424 sectors Units: sectors of 1 * 512 = 512 bytes Sector size (logical/physical): 512 bytes / 512 bytes I/O size (minimum/optimal): 512 bytes / 512 bytes Disklabel type: gpt Disk identifier: 98101B32-BBE2-4BF2-A06E-2BB33D000C20
> > > > > > > > > Device Start End Sectors Size Type /dev/mmcblk0p1 34 1569 1536 768K unknown /dev/mmcblk0p2 1570 2081 512 256K unknown /dev/mmcblk0p3 2082 2593 512 256K unknown /dev/mmcblk0p4 2594 6177 3584 1.8M unknown /dev/mmcblk0p5 6178 9761 3584 1.8M unknown /dev/mmcblk0p6 9762 10273 512 256K unknown /dev/mmcblk0p7 10274 10785 512 256K unknown /dev/mmcblk0p8 10786 11297 512 256K unknown /dev/mmcblk0p9 11298 11809 512 256K unknown /dev/mmcblk0p10 11810 12321 512 256K unknown /dev/mmcblk0p11 12322 12833 512 256K unknown /dev/mmcblk0p12 12834 13345 512 256K unknown /dev/mmcblk0p13 13346 14625 1280 640K unknown /dev/mmcblk0p14 14626 15905 1280 640K unknown /dev/mmcblk0p15 15906 16417 512 256K unknown /dev/mmcblk0p16 16418 28705 12288 6M unknown /dev/mmcblk0p17 28706 40993 12288 6M unknown /dev/mmcblk0p18 40994 4235297 4194304 2G unknown /dev/mmcblk0p19 4235298 4243489 8192 4M unknown /dev/mmcblk0p20 4243490 4366369 122880 60M unknown /dev/mmcblk0p21 4366370 4374561 8192 4M unknown /dev/mmcblk0p22 4374562 4415521 40960 20M unknown /dev/mmcblk0p23 4415522 4416545 1024 512K unknown /dev/mmcblk0p24 4416546 4595745 179200 87.5M unknown /dev/mmcblk0p25 4595746 5210145 614400 300M unknown /dev/mmcblk0p26 5210146 6258721 1048576 512M unknown /dev/mmcbl [... cut ...]  Down Speed :10 Half duplex eth0 PHY2 Down Speed :10 Half duplex eth0 PHY3 Down Speed :10 Half duplex eth0 PHY4 Down Speed :10 Half duplex eth0 PHY0 Down Speed :10 Half duplex eth0 PHY1 Down Speed :10 Half duplex eth0 PHY2 Down Speed :10 Half duplex eth0 PHY3 up Speed :10 Full duplex eth0 PHY4 Down Speed :10 Half duplex ipq6018_eth_init: done HTTP server is starting at IP: 192.168.1.1 done set host addr 0xa8c0 0x101 HTTP server is ready!
> 
> 然后依然ping不通192.168.1.1

看是没有识别到网线，还不知道是什么问题。重新插拔网线有用吗？

allonmymind, 2024-06-05:
> > > > > > > > > > > > > > > > 我也同样遇到问题，第一次刷您的uboot可以正常进web页面。一天后就进不去了，重刷也不行，但是刷旧的u-boot.mbn那个文件就可以正常打开，最开始我以为是我机器的问题，看到这个issue才知道原来不止我一个遇到了
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > 这不可能吧，刷进入uboot数据是不会变的，不会第一次可以隔一天不行。你刷uboot后还是使用原厂系统跑分吗？
> > > > > > > > > > > > > > 
> > > > > > > > > > > > > > 
> > > > > > > > > > > > > > 我是istore系统刷用ssh方法刷的您的uboot，应该是刷完以后改了大分区，不知道和这个有没有关系，现在进系统后日志会有这个报错： root@iStoreOS:~# dmesg | grep error [ 2.144729] UBIFS error (pid: 1): cannot open "ubi0:rootfs", error -19 [ 22.964140] Subsystem error monitoring/handling services are up [ 32.014082] Subsystem error monitoring/handling services are up [ 151.166555] Subsystem error monitoring/handling services are up [ 395.579422] Subsystem error monitoring/handling services are up [ 1140.617515] Subsystem error monitoring/handling services are up root@iStoreOS:~#
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 这些报错和uboot没啥关系，uboot是引导启动系统的。 我让群友测试，未发现亚瑟uboot进不了webui……不知道你们的是什么情况了。 我只有雅典娜，uboot正常使用。之前测试亚瑟uboot上一版，没什么问题，亚瑟已经出掉了…… 我想去JD买个亚瑟重新刷uboot看看了。
> > > > > > > > > > > > 
> > > > > > > > > > > > 
> > > > > > > > > > > > 感觉像是分区问题导致的，我的第一条报错也许和进不了uboot都是分区问题引起的
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > 用的是我教程中的分区表吗？提供下你用的分区表看看 不过感觉不应该，因为如果分区表有问题，系统是启动不了的
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > 是的，用的亚瑟2G那个 root@iStoreOS:~# blkid /dev/loop0: LABEL="rootfs_data" UUID="268e58b7-b101-4942-8fd0-4f32c07deb34" BLOCK_SIZE="4096" TYPE="ext4" /dev/mmcblk0p1: PARTLABEL="0:SBL1" PARTUUID="23691061-ce82-a7f9-e63d-92b9cdf584f6" /dev/mmcblk0p2: PARTLABEL="0:BOOTCONFIG" PARTUUID="89c1df88-fb18-5142-477d-de6878281347" /dev/mmcblk0p3: PARTLABEL="0:BOOTCONFIG1" PARTUUID="0802fa1e-92bf-ed7c-f7e0-20803765efd3" /dev/mmcblk0p4: PARTLABEL="0:QSEE" PARTUUID="0a81139c-3bb6-8db2-fae8-aaba889f4861" /dev/mmcblk0p5: PARTLABEL="0:QSEE_1" PARTUUID="4ba8c830-46ab-cf60-d7fc-2301436c266b" /dev/mmcblk0p6: PARTLABEL="0:DEVCFG" PARTUUID="6402e55f-c9f4-438b-7c1c-2bba6467f752" /dev/mmcblk0p7: PARTLABEL="0:DEVCFG_1" PARTUUID="a66ed3dd-0348-b9e1-58ab-2d10459cff71" /dev/mmcblk0p8: PARTLABEL="0:RPM" PARTUUID="084f94d8-8243-d676-2bcf-0b088dd6f58e" /dev/mmcblk0p9: PARTLABEL="0:RPM_1" PARTUUID="7296dd7d-817e-d7ac-b818-ab3100620029" /dev/mmcblk0p10: PARTLABEL="0:CDT" PARTUUID="60ba117e-3af5-3d6d-f41b-f90f8af17982" /dev/mmcblk0p11: PARTLABEL="0:CDT_1" PARTUUID="0b4dc452-ec49-8825-1b12-344036039c9e" /dev/mmcblk0p12: PARTLABEL="0:APPSBLENV" PARTUUID="65d77f3a-9863-97d7-0c00-a1826aa337ef" /dev/mmcblk0p13: PARTLABEL="0:APPSBL" PARTUUID="a656941f-4a5a-8e97-b038-3c56ca29cb26" /dev/mmcblk0p14: PARTLABEL="0:APPSBL_1" PARTUUID="ce2236d3-f9c8-9e3b-22d9-a0c67a008ea4" /dev/mmcblk0p15: PARTLABEL="0:ART" PARTUUID="ae4e3c02-5a5b-8f0d-f829-db226febe307" /dev/mmcblk0p16: PARTLABEL="0:HLOS" PARTUUID="1d04de0e-5189-61b6-57fb-b16554d1cdf3" /dev/mmcblk0p17: PARTLABEL="0:HLOS_1" PARTUUID="6c478b91-3026-043e-67b2-6bd87f42642f" /dev/mmcblk0p18: TYPE="squashfs" PARTLABEL="rootfs" PARTUUID="49ce1048-1499-ae66-4240-f15d706a2598" /dev/mmcblk0p19: PARTLABEL="0:WIFIFW" PARTUUID="4609454a-2ea6-cfa6-c005-531838befaf6" /dev/mmcblk0p20: PARTLABEL="rootfs_1" PARTUUID="3db96119-20c1-90b8-3dd6-43f241ac87b8" /dev/mmcblk0p21: PARTLABEL="0:WIFIFW_1" PARTUUID="59e7b7d5-03c4-caba-d2e8-3661ff6ab7bd" /dev/mmcblk0p22: PARTLABEL="rootfs_data" PARTUUID="87d7a95d-007b-f9d0-3c54-dd4f9140d35e" /dev/mmcblk0p23: PARTLABEL="0:ETHPHYFW" PARTUUID="c03e0f0b-3f28-ec31-0c36-d119817baeb0" /dev/mmcblk0p24: PARTLABEL="plugin" PARTUUID="260f092b-3b3a-3710-4f48-846ec245d3a1" /dev/mmcblk0p25: PARTLABEL="log" PARTUUID="837fae9b-8118-b2ec-f195-877a5a0133cd" /dev/mmcblk0p26: PARTLABEL="swap" PARTUUID="7f63406e-c870-c8b0-489a-45cdec955611" /dev/mmcblk0p27: UUID="e9f9ad5b-8bda-4c1a-a9eb-1bbad037044c" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="cc852b77-5469-4548-9c36-9e3ec24afd0e" root@iStoreOS:~# fdisk -l /dev/mmcblk0 Disk /dev/mmcblk0: 57.12 GiB, 61329113088 bytes, 119783424 sectors Units: sectors of 1 * 512 = 512 bytes Sector size (logical/physical): 512 bytes / 512 bytes I/O size (minimum/optimal): 512 bytes / 512 bytes Disklabel type: gpt Disk identifier: 98101B32-BBE2-4BF2-A06E-2BB33D000C20
> > > > > > > > > > Device Start End Sectors Size Type /dev/mmcblk0p1 34 1569 1536 768K unknown /dev/mmcblk0p2 1570 2081 512 256K unknown /dev/mmcblk0p3 2082 2593 512 256K unknown /dev/mmcblk0p4 2594 6177 3584 1.8M unknown /dev/mmcblk0p5 6178 9761 3584 1.8M unknown /dev/mmcblk0p6 9762 10273 512 256K unknown /dev/mmcblk0p7 10274 10785 512 256K unknown /dev/mmcblk0p8 10786 11297 512 256K unknown /dev/mmcblk0p9 11298 11809 512 256K unknown /dev/mmcblk0p10 11810 12321 512 256K unknown /dev/mmcblk0p11 12322 12833 512 256K unknown /dev/mmcblk0p12 12834 13345 512 256K unknown /dev/mmcblk0p13 13346 14625 1280 640K unknown /dev/mmcblk0p14 14626 15905 1280 640K unknown /dev/mmcblk0p15 15906 16417 512 256K unknown /dev/mmcblk0p16 16418 28705 12288 6M unknown /dev/mmcblk0p17 28706 40993 12288 6M unknown /dev/mmcblk0p18 40994 4235297 4194304 2G unknown /dev/mmcblk0p19 4235298 4243489 8192 4M unknown /dev/mmcblk0p20 4243490 4366369 122880 60M unknown /dev/mmcblk0p21 4366370 4374561 8192 4M unknown /dev/mmcblk0p22 4374562 4415521 40960 20M unknown /dev/mmcblk0p23 4415522 4416545 1024 512K unknown /dev/mmcblk0p24 4416546 4595745 179200 87.5M unknown /dev/mmcblk0p25 4595746 5210145 614400 300M unknown /dev/mmcblk0p26 5210146 [... cut ...] PHY2 Down Speed :10 Half duplex eth0 PHY3 Down Speed :10 Half duplex eth0 PHY4 Down Speed :10 Half duplex eth0 PHY0 Down Speed :10 Half duplex eth0 PHY1 Down Speed :10 Half duplex eth0 PHY2 Down Speed :10 Half duplex eth0 PHY3 up Speed :10 Full duplex eth0 PHY4 Down Speed :10 Half duplex ipq6018_eth_init: done HTTP server is starting at IP: 192.168.1.1 done set host addr 0xa8c0 0x101 HTTP server is ready!
> > 然后依然ping不通192.168.1.1
> 
> 看是没有识别到网线，还不知道是什么问题。重新插拔网线有用吗？ 这里回复太多了，你可以加Q群私聊837365344

已加

rooneysh, 2024-06-06:
Same issue found on my device.
AX1800 Pro
JDC02-1.3.0.r2106

After flash uboot and mmcblk0, can't enter uboot page.
Can still access 192.168.68.1, WiFi couldn't enable also.
root password had been set to 'admin'.

ldd200888, 2024-06-06:
看ttl输出，感觉像是端口速率的原因，一直在设置速率。。。
是不是设置为10M 半双工就能访问了？

@allonmymind @lgs2007m

lgs2007m, 2024-06-07:
@ldd200888 @allonmymind @rooneysh @weah2000 我买了两台京东云AX1800 Pro进行测试，未能复现你们的问题，没办法了。

一台全新64G亚瑟，2023.04出厂，固件r2242，按教程刷了uboot 20240510后两台电脑分别测试了5次进uboot，浏览器chrome、edge，都没问题。也刷了512M、1024M、2048M分区表，都可以进uboot。
一台5月份开始跑分的准新128G亚瑟，2022.04出厂，固件r2080，按教程刷了uboot 20240510后电脑1测试了5次进uboot，浏览器chrome、edge，都没问题。也刷了512M、1024M、2048M分区表，都可以进uboot。

电脑1是Intel NUC11 网卡Intel I225-V 系统Win11
电脑2是宏碁AS5750G 网卡BCM57785 系统Win7

ldd200888, 2024-06-08:
@allonmymind 这个不是开了ttl了么？也没找出来原因是啥么？

另外，看下你的连接uboot后，协商的速率是啥？

rooneysh, 2024-06-10:
> 看ttl输出，感觉像是端口速率的原因，一直在设置速率。。。 是不是设置为10M 半双工就能访问了？
> 
> @allonmymind @lgs2007m

Thanks for the information.
I changed my computer and ethernet cable another day, and it's is ok now.
I think it's related to RJ45 connector type.

lgs2007m, 2024-06-10:
> > 看ttl输出，感觉像是端口速率的原因，一直在设置速率。。。 是不是设置为10M 半双工就能访问了？
> > @allonmymind @lgs2007m
> 
> Thanks for the information. I changed my computer and ethernet cable another day, and it's is ok now. I think it's related to RJ45 connector type.

Thanks for your information.

ldd200888, 2024-06-10:
> > 看ttl输出，感觉像是端口速率的原因，一直在设置速率。。。 是不是设置为10M 半双工就能访问了？
> > @allonmymind @lgs2007m
> 
> Thanks for the information. I changed my computer and ethernet cable another day, and it's is ok now. I think it's related to RJ45 connector type.

good news! @lgs2007m  是不是能针对这个修复下？

lgs2007m, 2024-06-10:
> > > 看ttl输出，感觉像是端口速率的原因，一直在设置速率。。。 是不是设置为10M 半双工就能访问了？
> > > @allonmymind @lgs2007m
> > 
> > 
> > Thanks for the information. I changed my computer and ethernet cable another day, and it's is ok now. I think it's related to RJ45 connector type.
> 
> good news! @lgs2007m 是不是能针对这个修复下？

我复现不了问题，无从下手，不懂怎么弄

laipeng668, 2024-06-18:
> > > > 看ttl输出，感觉像是端口速率的原因，一直在设置速率。。。 是不是设置为10M 半双工就能访问了？
> > > > @allonmymind @lgs2007m
> > > 
> > > 
> > > Thanks for the information. I changed my computer and ethernet cable another day, and it's is ok now. I think it's related to RJ45 connector type.
> > 
> > 
> > good news! @lgs2007m 是不是能针对这个修复下？
> 
> 我复现不了问题，无从下手，不懂怎么弄

> > > > 看ttl输出，感觉像是端口速率的原因，一直在设置速率。。。 是不是设置为10M 半双工就能访问了？
> > > > @allonmymind @lgs2007m
> > > 
> > > 
> > > Thanks for the information. I changed my computer and ethernet cable another day, and it's is ok now. I think it's related to RJ45 connector type.
> > 
> > 
> > good news! @lgs2007m 是不是能针对这个修复下？
> 
> 我复现不了问题，无从下手，不懂怎么弄

我今天才按你的教程从头到尾刷了一遍，刷了GPT分区表后，从路由器的灯（红灯闪几次后变蓝灯）判断，已经是在uboot状态了，但是uboot的web界面一直进不去。我发现即使我指定电脑ip，网络状态还是无法连接网络，我只能尝试其他方式。后来只要把wan口的网线拔掉（之前是连接上级路由的），就可以获取网络，并且很顺利的进入uboot的web界面了，期间还禁用后又启用了网卡。uboot是没问题的，可能是wan口接了其他设备，有冲突导致不能进入web界面。

非常感谢你提供的uboot和刷机教程！

lgs2007m, 2024-06-18:
> 我今天才按你的教程从头到尾刷了一遍，刷了GPT分区表后，从路由器的灯（红灯闪几次后变蓝灯）判断，已经是在uboot状态了，但是uboot的web界面一直进不去。我发现即使我指定电脑ip，网络状态还是无法连接网络，我只能尝试其他方式。后来只要把wan口的网线拔掉（之前是连接上级路由的），就可以获取网络，并且很顺利的进入uboot的web界面了，期间还禁用后又启用了网卡。uboot是没问题的，可能是w [...]

感谢提供信息！

ldd200888, 2024-06-19:
我进不去的时候，wan口是没有接网线的。

taotieren, 2024-06-21:
我这边也用雅典娜拆机接夹串口刷 `uboot-JDC_AX1800_Pro-AX6600_Athena-20240510.bin` 固件验证了下

```bash
系统是 Arch Linux
串口是 CH340N
主机网口是  2.5G
网线 五类不带屏蔽、六类单层屏蔽线、七类双层屏蔽 
主机 IP 192.168.1.1
```
刷完之后遇到同样的问题主机 IP 192.168.1.1/24 连接 LAN 口四个之一
会出现雅典娜 LAN 绿灯闪几下，主机提示网络已连接然后断开， 雅典娜 LAN 灯熄灭，换其他 四个口同样问题，
网线从五类换到七类也是同样现象。
手动修改主机网络协商速率为 `1Gbit/s` 后再拔掉网线重新插入另外 LAN 口上，提示连接成功。
通过串口看日志是识别成 `1000Mbit/s` 用官方 uboot 也是同样的问题，也需要手动修改主机的连接速率为 `1Gbit/s`。

taotieren, 2024-06-21:
`uboot-JDC_AX1800_Pro-AX6600_Athena-20240510.bin` 刷进 雅典娜路由里面，在雅典娜 uboot web 界面能刷 亚瑟 OP 固件吗？
刷了 雅典娜 2048M 分区表，eMMC 剩余分区无法识别，是要刷其他固件进 ssh 后分区吗？

[20 more replies]

closed 2024-08-16 by lgs2007m