lgs2007m/Actions-OpenWrt #20: 如果百里想彻底刷回原厂固件，是不是直接把之前备份的bin文件全部重新dd回就行了，还有分区表

itq5, 2024-06-07:
如果百里想彻底刷回原厂固件，是不是直接把之前备份的bin文件全部重新dd回就行了，还有分区表

lgs2007m, 2024-06-07:
不需要，具体看我的教程吧。
不要教程都不看就提问，浪费我的时间

closed 2024-06-07 by lgs2007m

OpenSourceUser2021, 2024-07-05:
借道问下，如果要彻底回到原厂数据 包括UBOOT,BL,FIP 和GPT 等，要如何操作呢？ ，之前已按教程流程备份数据。

lgs2007m, 2024-07-05:
> 借道问下，如果要彻底回到原厂数据 包括UBOOT,BL,FIP 和GPT 等，要如何操作呢？ ，之前已按教程流程备份数据。

删除环境变量：
fw_setenv jdc_crc_version
fw_setenv jdc_opp_version

进uboot http://192.168.1.1/gpt.html 刷备份的原厂gpt即mmcblk0p1_PMBR.bin文件，会自动重启，再进uboot http://192.168.1.1 刷小于52M的OP固件
系统启动后，先保存下分区表，然后查看下分区表：
echo -e 'w' | fdisk /dev/mmcblk0
fdisk -l /dev/mmcblk0

格式化最后一个分区：
mkfs.ext4 $(blkid -t PARTLABEL=storage -o device)

挂载分区：
mkdir /mnt/mmcblk0p13
mount -t ext4 /dev/mmcblk0p13 /mnt/mmcblk0p13

上传log、plugin、swap备份到/mnt/mmcblk0p13并刷回：
dd if=/mnt/mmcblk0p13/mmcblk0p10_log.bin of=$(blkid -t PARTLABEL=log -o device) conv=fsync
dd if=/mnt/mmcblk0p13/mmcblk0p11_plugin.bin of=$(blkid -t PARTLABEL=plugin -o device) conv=fsync
dd if=/mnt/mmcblk0p13/mmcblk0p12_swap.bin of=$(blkid -t PARTLABEL=swap -o device) conv=fsync

设置系统0启动并恢复kernel、rootfs和清零rootfs_data：
fw_setenv dual_boot.current_slot 0
dd if=/tmp/mmcblk0p5_kernel.bin of=$(blkid -t PARTLABEL=kernel -o device)
dd if=/tmp/mmcblk0p6_rootfs.bin of=$(blkid -t PARTLABEL=rootfs -o device)
dd if=/tmp/mmcblk0p5_kernel.bin of=$(blkid -t PARTLABEL=kernel2 -o device)
dd if=/tmp/mmcblk0p6_rootfs.bin of=$(blkid -t PARTLABEL=rootfs2 -o device)
dd if=/dev/zero of=$(blkid -t PARTLABEL=rootfs_data -o device)

没报错则断电重启进原厂系统，系统没问题再uboot刷小于52MB的OP固件，进系统恢复原厂uboot和bl2：
ash
echo 0 > /sys/block/mmcblk0boot0/force_ro
dd if=/tmp/mmcblk0boot0_bl2.bin of=/dev/mmcblk0boot0 conv=fsync
echo 1 > /sys/block/mmcblk0boot0/force_ro
dd if=/tmp/mmcblk0p4_fip.bin of=$(blkid -t PARTLABEL=fip -o device) conv=fsync

刷uboot和bl2很重要，要验证md5和备份的文件是否一样，否则不能重启：
md5sum /dev/mmcblk0boot0 && md5sum $(blkid -t PARTLABEL=fip -o device)

再设置系统0启动并恢复kernel、rootfs和清零rootfs_data：
fw_setenv dual_boot.current_slot 0
dd if=/tmp/mmcblk0p5_kernel.bin of=$(blkid -t PARTLABEL=kernel -o device)
dd if=/tmp/mmcblk0p6_rootfs.bin of=$(blkid -t PARTLABEL=rootfs -o device)
dd if=/tmp/mmcblk0p5_kernel.bin of=$(blkid -t PARTLABEL=kernel2 -o device)
dd if=/tmp/mmcblk0p6_rootfs.bin of=$(blkid -t PARTLABEL=rootfs2 -o device)
dd if=/dev/zero of=$(blkid -t PARTLABEL=rootfs_data -o device)

没报错断电重启即可，然后进手机app设置内置存储为本地网盘，然后直接恢复出厂，启动后再设置内置存储为智能加速服务。

OpenSourceUser2021, 2024-07-05:
> 删除环境变量： fw_setenv jdc_crc_version fw_setenv jdc_opp_version
>
> 进uboot http://192.168.1.1/gpt.html 刷备份的原厂gpt即mmcblk0p1_PMBR.bin文件，会自动重启，再进uboot http://192.168.1.1 刷小于52M的OP固件 系统启动后，先保存下分区表，然后查看下分区表： [...]

非常感谢，留作备用，以防出问题需要找JD 售后

lgs2007m, 2024-10-14:
> @lgs2007m 你好，请教一下上面的回复里 `刷小于52M的OP固件` ，我找了好几个固件都刷不进去。你 Actions 的固件又太大，我应该去哪里找一个小于52M的固件？ 还是说只能 fork 项目编译一个了。
> 
> ~试过 QWRT ，试过主线固件，试过压缩包的op过渡固件，刷了备份分区再刷固件后就无限进uboot了~
> 
> 经测试 QWRT 是可以的，绿灯之后ssh登录就可以。
> 
> 来自恩山 lean 的[固件](https://pan.baidu.com/s/1n7eJbRg6lt2JJnrvllRSdg?pwd=mn7w) ( 固件超过25M上传不到 GitHub )

为什么要全部分区刷回原厂？你要返厂保修吗？不是的话，这样做就是折腾自己，浪费生命
现在教程的分区、uboot就支持原厂跑分，只需要把最后几个分区还原即可。
小固件可以下载：
https://github.com/VIKINGYFY/CloseWRT-CI/releases/tag/MTK-7986_hanwckf-openwrt-21.02_24.10.14_05.57.17