lgs2007m/Actions-OpenWrt #22: 百里Docker问题请教

peplemon, 2024-06-09:
猫大，端午安康。
1.您教程里面说：“如果要把storage分区挂载给docker，则在系统->挂载点菜单，添加挂载点，UUID选择mmcblk0p9，输入自定义挂载位置/opt”，我看了一下，已挂载的文件系统里面已经有1个1G的/opt挂载点，指向的是overlayfs:/overlay。两个同名的/opt挂载点会不会冲突呢，需要把1G的挂载点卸载分区吗?
2.昨天自学编译了hanwckf大佬的固件，docker版本是20.10.26，这个版本可以正常使用吗？我测试没有成功，可能是我自己的问题
3.前几天问题您给closed了，我反馈一下测试效果
① CPU占用问题，测试了lean的qwrt，占用不高，我自己编译的固件（基于hanwckf，插件较少，占用也不高，基本在0-5%之间波动）
② 插件问题，自己编译了ssrp，可以添加naive协议。改源后刷新虽然可以看到新的插件版本（比如docker可以看到26版），但由于缺少各种依赖，直接升级失败。目前个人感觉，只要插件稳定能用就行，没必要追求最新版。

lgs2007m, 2024-06-09:
1. 直接挂载最后一个大分区为/opt即可，不需要卸载其他的。因为不挂载的话默认使用overlay最为docker存储空间，固件没有集成fuse-overlayfs，docker是不支持在overlay存储的。overlay太小，而且uboot刷固件会清除overlay，也不建议docker使用overlay。
2. hanwckf大佬源码的软件feeds源使用Immortalwrt openwrt-21.02分支软件源，所以版本有点旧。我的docker也是20.10.26，使用上未发现问题。需要新版可以删除Immortalwrt的软件源里面的docker，自己替换新版的进去。
3. 占用问题可能是autocore显示问题或者某个插件的问题吧，具体也不太懂。插件需要新的尝试换源，或者删除update feeds后删除旧的替换新的，不过可能有兼容问题，需要不断尝试……一般插件都可以直接安装，但是缺少kmod依赖的就只能重新编译了，因为每个仓库编译出来的kmod都不同，不兼容。我能力有限，感兴趣只能你自己多尝试了

issue关闭了，也可以回复的，不要紧

peplemon, 2024-06-09:
谢谢猫大
您看这么挂载可以吗？（昨天docker错误可能是没挂载的原因）
[image]

docker根目录如下
[image]

lgs2007m, 2024-06-09:
挂载之后重启系统下，已挂载文件系统那个overlayfs挂载/opt应该就不在，overlayfs挂载/opt不在才正常。
docker根目录下那个路径默认就是这样的，没问题。

peplemon, 2024-06-09:
猫大，启用勾选了，保存应用，然后重启了，怎么上面哪个/opt还在呢？删掉挂载了两边，重启了两遍。好奇怪啊

[image]

lgs2007m, 2024-06-09:
额，你可以看看docker配置那里，存储空间有多大，如果是53G那就没错了，如果不是就有问题。
可以尝试重置系统，重新配置这个挂载点。

我这边重置系统后挂载是这样的：
[image]

挂载最后一个大分区为/opt后重启是这样的：
[image]
已挂载文件系统最下面四个overlay是四个docker镜像。

peplemon, 2024-06-09:
猫大，按您说的重置系统，重新挂载，重启，还是不行
能否给mmcblk0p9挂在成其他名字，然后给docker根目录指定这个名字？

lgs2007m, 2024-06-09:
不需要吧，直接看docker概览那里Docker根目录存储空间正常就行。直接用，有问题再说。
其他挂载没试过，需要你自行搜索下

peplemon, 2024-06-09:
猫大，docker根目录空间不正常，只显示1点几G
然后我尝试将mmcblk0p9挂载成ppt的名字，保存确认后，上面那个小分区竟然跟着改名了，两个挂载点是联动的
[image]
[image]

lgs2007m, 2024-06-09:
不知道这个是什么问题……你可以刷我编译的固件试试。

peplemon, 2024-06-09:
好的，谢谢猫大，我再研究研究

peplemon, 2024-06-09:
猫大，突然想起来，这个storage分区只用教程中的命令格式化了一次，后面再刷固件就没格式化过了，是否每次刷固件都要格式化一次呢？

lgs2007m, 2024-06-09:
不需要。只有改分区表，最后一个分区才需要格式化一次。
如果你放心，也可以重新格式化一次。后面不需要格式化了

peplemon, 2024-06-09:
猫大，实在是搞定不了。有个不情之请，您下次编译的时候，能否
1. 把ssrp插件里面的协议全选一下，现在的固件里面协议不全
2. 添加一下mosdns
以后直接用猫大的固件了
这个不着急，您下次更新固件时候考虑下就行，我已经放弃编译了。

lgs2007m, 2024-06-09:
我那个固件也不能挂载/opt吗？
可以的话你可以直接改我Actions百里的config，加下协议和mosdns就行了，或者直接安装看看。
因为我使用已经稳定了，最近没什么心思搞这个，不知道下次什么搞了

peplemon, 2024-06-09:
猫大，如果直接使用你这个[.mtwifi-cfg.config]，那还用H大的编译源包吗，还是全盘用你的，我是本地ubuntu编译的

lgs2007m, 2024-06-09:
是的，都是H大。现在闭源的都是H大的源码编译的
你把本地编译加的ssrp协议加进去就行了，mosdns直接安装看看，下载下面三个文件，然后上传按顺序安装
https://github.com/sbwml/luci-app-mosdns/releases
mosdns_5.3.1-1_aarch64_cortex-a53.ipk
luci-app-mosdns_1.5.20_all.ipk
luci-i18n-mosdns-zh-cn_git-23.319.54348-e79a991_all.ipk

peplemon, 2024-06-09:
猫大，你的config文件怎么是x86的呢？
[image]

lgs2007m, 2024-06-09:
你要将.mtwifi-cfg.config文件改名.config放进去

peplemon, 2024-06-09:
是这么操作的，进去界面如下，还要再选择一遍吗？比如平台，设备型号等

[image]

lgs2007m, 2024-06-09:
检查.config放到immortalwrt-mt798x根目录下了吗？

peplemon, 2024-06-09:
放到了啊，那个config.ini是自带的
[image]
[image]

lgs2007m, 2024-06-09:
那这不可能呀，我从来没遇到过。关闭当前终端，重新来过

peplemon, 2024-06-09:
猫大，你传个config文件给我吧，我感觉自己下载的可能不对
终端重开了，还是x86

lgs2007m, 2024-06-09:
[mtwifi-cfg.zip](https://github.com/user-attachments/files/15751156/mtwifi-cfg.zip)
.config是个文件文件，直接打开就知道对不对了，也可以直接复制文本粘贴进去就行

closed 2024-06-09 by lgs2007m