lgs2007m/Actions-OpenWrt #23: 请问大佬，刷机后2.5G口是在WAN还是LAN？

hereVg0, 2024-06-10:
我的宽带是1000兆，今天测速的时候发现（光猫桥接，百里拨号，连wifi测），怎么都达不到，最多也就970M。
如果用光猫拨号，连光猫的wifi测速，是能达到1200兆左右的。
请问固件默认是WAN口2.5G吗？为什么百里拨号达不到1000兆？还是说默认LAN口是2.5G，那么请问该如何切换呢？。感谢大佬

lgs2007m, 2024-06-10:
没有千兆宽带，不清楚拨号的情况。我只试过wan口使用DHCP模式，接2.5G路由器的测速服务，再测WiFi是可以到1200M的。

hereVg0, 2024-06-10:
默认2.5G口是在wan还是lan呀？怎么切换？

lgs2007m, 2024-06-10:
默认是WAN，我也不懂切换，自行搜索设置，这里不做使用指导。

closed 2024-06-10 by lgs2007m

wzcsimon123, 2024-06-26:
把eth1 放到br-lan中 把lan1当wan口就行了

lantinglou, 2024-07-20:
Lan1怎么设置为WAN呢

lantinglou, 2024-07-20:
> 把eth1 放到br-lan中 把lan1当wan口就行了

ethi 放入br-lan可以了，但是lan1怎么设置wan 没找到对应设置的地方。 谢谢

lgs2007m, 2024-08-16:
双wan设置参考237大佬的教程https://www.right.com.cn/forum/thread-8316313-1-1.html
[mt798x双wan设置.pdf](https://github.com/user-attachments/files/16632968/mt798x.wan.pdf)

最近需要用路由器2.5G做测速服务，改了下wan为lan，靠近wan的lan1改为wan，你可以参考看看
[image]

修改后的/etc/config/network文件中的mac是我自己的，你改为你的就行，改完reboot重启路由器
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'
	list ports 'eth1'

config device
	option name 'eth0'
	option macaddr 'dc:d8:7c:55:55:50'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'eth0.2'
	option proto 'dhcp'
	option macaddr 'dc:d8:7c:55:55:51'

config interface 'wan6'
	option device 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '2 3 4 6t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '1 6t'

config interface 'docker'
	option device 'docker0'
	option proto 'none'
	option auto '0'

config device
	option type 'bridge'
	option name 'docker0'

```