lgs2007m/Actions-OpenWrt #24: 老哥，你写的太好了，但对于纯白来说还是视频来的好

dcvh556tfg, 2024-06-13:
我说真的，写的很详细，没有别的意思，小白的我还是搞不好，要有视频教程就好了

lgs2007m, 2024-06-14:
需要有基础，不是面向小白的。
不做宣传，不会做视频了。

closed 2024-06-14 by lgs2007m

dcvh556tfg, 2024-06-14:
> 需要有基础，不是面向小白的。 不做宣传，不会做视频了。

那有没有推荐的京东云亚瑟固件，我刷了 5 天了，一直没找到好用的，就纯粹想搞一个正常使用 docker 的

lgs2007m, 2024-06-14:
> > 需要有基础，不是面向小白的。 不做宣传，不会做视频了。
> 
> 那有没有推荐的京东云亚瑟固件，我刷了 5 天了，一直没找到好用的，就纯粹想搞一个正常使用 docker 的

最好的还是QWRT，自己安装docker，可以上恩山找免费版

dcvh556tfg, 2024-06-14:
> > > 需要有基础，不是面向小白的。 不做宣传，不会做视频了。
> > 
> > 
> > 那有没有推荐的京东云亚瑟固件，我刷了 5 天了，一直没找到好用的，就纯粹想搞一个正常使用 docker 的
> 
> 最好的还是QWRT，自己安装docker，可以上恩山找免费版

老哥，我已经试过了，lean 的 openwrt 还有好些叫不上名字的，安装 docker 有的报错，有的更新软件源就报错，还有的差一点就能下载镜像就是卡死了实在找不到好的了

KunYi, 2024-06-15:
> > > > 需要有基础，不是面向小白的。 不做宣传，不会做视频了。
> > > 
> > > 
> > > 那有没有推荐的京东云亚瑟固件，我刷了 5 天了，一直没找到好用的，就纯粹想搞一个正常使用 docker 的
> > 
> > 
> > 最好的还是QWRT，自己安装docker，可以上恩山找免费版
> 
> 老哥，我已经试过了，lean 的 openwrt 还有好些叫不上名字的，安装 docker 有的报错，有的更新软件源就报错，还有的差一点就能下载镜像就是卡死了实在找不到好的了

猜測沒改分區表，分區表沒改就會這樣，如果你分區表有改成功的話就當我沒說