lgs2007m/Actions-OpenWrt #27: JDC-AX1800 Pro亚瑟更新固件r4211后还能TTL刷机吗

xcf13363175, 2024-06-25:
请教，JDC-AX1800 Pro亚瑟更新固件r4211后还能TTL刷机吗，现在不敢升级。

lgs2007m, 2024-06-25:
不清楚，目前只见到恩山有一个人开着ssh跑原厂系统，被监测到，然后强制更新了uboot，新uboot不能中断进控制台了。
初步怀疑是个例，因为我直接从旧固件升级到4.0.0.r4015，试了几次，uboot没有被更新。所以不建议原厂固件开ssh后联网。
如果当前可以ssh，建议先断网，开ssh后先刷个不死uboot，然后重置系统，再联网去更新固件。

lgs2007m, 2024-06-25:
最新的固件的确包含uboot，可能会强制更新uboot，不建议升级！

taotieren, 2024-07-02:
我这边 uboot 也被覆盖了，现在也进去不了 uboot 了

taotieren, 2024-07-02:
是刚升级仓库的最新的固件发现进不了 uboot 了。
在 OP 中格式化的 27 分区现在在 雅典娜中也识别不了了

```bash
683.263176] EXT4-fs (mmcblk0p27): couldn't mount RDWR because of unsupported optional features (10000)
```

taotieren, 2024-07-02:
这是新固件的串口日志

```bash

Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.XF.0.3-00100-IPQ60xxLZB-1
S - IMAGE_VARIANT_STRING=IPQ6018LA
S - OEM_IMAGE_VERSION_STRING=crm-ubuntu124
S - Boot Interface: eMMC
S - Secure Boot: Off
S - Boot Config @ 0x000a602c = 0x000002e2
S - JTAG ID @ 0x000a607c = 0x0013a0e1
S - OEM ID @ 0x000a6080 = 0x00000000
S - Serial Number @ 0x000a4128 = 0x4eabfb42
S - OEM Config Row 0 @ 0x000a4188 = 0x0000000000000000
S - OEM Config Row 1 @ 0x000a4190 = 0x0000000000000000
S - Feature Config Row 0 @ 0x000a4130 = 0x0000000008000001
S - Feature Config Row 1 @ 0x000a4138 = 0x02c3e83383000009
S - PBL Patch Ver: 1
S - I-cache: On
S - D-cache: On
B -      3413 - PBL, Start
B -       592 - bootable_media_detect_entry, Start
B -      4339 - bootable_media_detect_success, Start
B -    142551 - elf_loader_entry, Start
B -    142722 - auth_hash_seg_entry, Start
B -    143864 - auth_hash_seg_exit, Start
B -    144356 - elf_segs_hash_verify_entry, Start
B -    170847 - elf_segs_hash_verify_exit, Start
B -    175036 - auth_xbl_sec_hash_seg_entry, Start
B -    175180 - auth_xbl_sec_hash_seg_exit, Start
B -    181729 - xbl_sec_segs_hash_verify_entry, Start
B -    181730 - xbl_sec_segs_hash_verify_exit, Start
B -    182659 - PBL, End
B -    149846 - SBL1, Start
B -    292769 - GCC [RstStat:0x0, RstDbg:0x600000] WDog Stat : 0x4
B -    295209 - clock_init, Start
D -      3690 - clock_init, Delta
B -    304756 - boot_flash_init, Start
D -     53131 - boot_flash_init, Delta
B -    360388 - sbl1_ddr_set_default_params, Start
D -       274 - sbl1_ddr_set_default_params, Delta
B -    366823 - boot_config_data_table_init, Start
D -      2257 - boot_config_data_table_init, Delta - (575 Bytes)
B -    375912 - CDT Version:2,Platform ID:8,Major ID:3,Minor ID:2,Subtype:2
B -    381494 - Image Load, Start
D -      6588 - OEM_MISC Image Loaded, Delta - (0 Bytes)
B -    390827 - Image Load, Start
D -      5063 - PMIC Image Loaded, Delta - (0 Bytes)
B -    398696 - sbl1_ddr_set_params, Start
B -    403698 - CPR configuration: 0x555
B -    406900 - Pre_DDR_clock_init, Start
D -       213 - Pre_DDR_clock_init, Delta
D -         0 - sbl1_ddr_set_params, Delta
B -    442372 - Image Load, Start
D -       518 - APDP Image Loaded, Delta - (0 Bytes)
B -    461068 - Image Load, Start
D -       519 - QTI_MISC Image Loaded, Delta - (0 Bytes)
B -    463752 - Image Load, Start
D -       854 - Auth Metadata
D -       671 - Segments hash check
D -     11773 - QSEE Dev Config Image Loaded, Delta - (37154 Bytes)
B -    477416 - Image Load, Start
D -      6527 - Auth Metadata
D -     10370 - Segments hash check
D -     94367 - QSEE Image Loaded, Delta - (1439116 Bytes)
B -    572302 - Image Load, Start
D -       793 - Auth Metadata
D -       946 - Segments hash check
[... 449 lines cut ...]
[   15.124286] col:14 time:200 36
[   15.324282] col:14 time:200 36
[   15.521430] ssdk_switch_register[1718]:INFO:Chip version 0x1501
[   15.521462] qca_link_polling_select[1315]:INFO:link-polling-required node does not exist
[   15.524282] col:12 time:200 36
[   15.534476] ssdk_switch_register[1744]:INFO:polling is selected
[   15.537341] regi_init[3571]:INFO:Initializing HPPE Done!!
[   15.543200] regi_init[3631]:INFO:qca-ssdk module init succeeded!
[   15.549693] tun: Universal TUN/TAP device driver, 1.6
[   15.554805] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[   15.562707] EDMA ver 1 hw init
[   15.566054] EDMA HW Reset completed succesfully
[   15.568877] Num rings - TxDesc:1 (23-23) TxCmpl:1 (23-23)
[   15.573276] RxDesc:1 (15-15) RxFill:1 (7-7)
[   15.679832] scsi 0:0:0:0: Direct-Access     ZHITAI S C001 Active      ZT01 PQ: 0 ANSI: 6
[   15.681213] sd 0:0:0:0: [sda] 1000215216 512-byte logical blocks: (512 GB/477 GiB)
[   15.687773] sd 0:0:0:0: [sda] Write Protect is off
[   15.695153] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   15.703985] sd 0:0:0:0: [sda] Attached SCSI disk
[   15.724297] col:12 time:200 36
[   15.784769] **********************************************************
[   15.784793] * NSS Data Plane driver
[   15.790180] **********************************************************

```

taotieren, 2024-07-02:
uboot 中的串口接收被禁用了，在 minicom 中操作键盘是无反应，不排除是改了接收格式

lgs2007m, 2024-07-03:
> 是刚升级仓库的最新的固件发现进不了 uboot 了。 在 OP 中格式化的 27 分区现在在 雅典娜中也识别不了了
> 
> ```shell
> 683.263176] EXT4-fs (mmcblk0p27): couldn't mount RDWR because of unsupported optional features (10000)
> ```

不能在线升级，可uboot直接刷最新固件。最新固件还没有破解telnet的方法放出，你现在uboot已经不能中断了，拆机用USB 9008救砖，重刷uboot吧

taotieren, 2024-07-03:
> > 是刚升级仓库的最新的固件发现进不了 uboot 了。 在 OP 中格式化的 27 分区现在在 雅典娜中也识别不了了
> > ```shell
> > 683.263176] EXT4-fs (mmcblk0p27): couldn't mount RDWR because of unsupported optional features (10000)
> > ```
> 
> 不能在线升级，可uboot直接刷最新固件。最新固件还没有破解telnet的方法放出，你现在uboot已经不能中断了，拆机用USB 9008救砖，重刷uboot吧

找到 qsaharaserver 的源码仓库了：https://github.com/Muhmmad-Almuhmmah/QSaharaServer/tree/main
顺便编译了一个 linux 版本出来了

Arch 用户可以通过 AUR 安装，
其他 linux 可以通过解压下面压缩包使用

[qsaharaserver.tar.gz](https://github.com/user-attachments/files/16086235/qsaharaserver.tar.gz)

lgs2007m, 2024-07-03:
> 找到 qsaharaserver 的源码仓库了：https://github.com/Muhmmad-Almuhmmah/QSaharaServer/tree/main 顺便编译了一个 linux 版本出来了
>
> Arch 用户可以通过 AUR 安装， 其他 linux 可以通过解压下面压缩包使用 [...]

你不需要搞这些，直接下载我的刷机教程，看里面的USB救砖即可，所有文件都有，还有个Windows用bat脚本

taotieren, 2024-07-03:
我这边跑得 Linux 系统，跑 win 脚本我还要装个 win 系统。

taotieren, 2024-07-03:
短接后进入，usb 能识别到
```bash
❯ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1a81:2053 Holtek Semiconductor, Inc. ikbc d4b
Bus 001 Device 003: ID 34d1:0608  USB2.0 HUB
Bus 001 Device 004: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 005: ID 1d57:fa61 Xenta 2.4G Wireless Mouse
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 2b7e:b743 Kingcome HD Webcam
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 003: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 007 Device 006: ID 1a86:8091 QinHeng Electronics USB HUB
Bus 007 Device 007: ID 05e3:0751 Genesys Logic, Inc. microSD Card Reader
Bus 007 Device 008: ID 0c76:153f JMTek, LLC. USB PnP Audio Device
Bus 007 Device 010: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 007 Device 017: ID 1a86:55da QinHeng Electronics USB Dual_Serial
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 002: ID 05e3:0626 Genesys Logic, Inc. Hub
Bus 008 Device 003: ID 0bda:8156 Realtek Semiconductor Corp. USB 10/100/1G/2.5G LAN
Bus 008 Device 004: ID 05e3:0616 Genesys Logic, Inc. hub
❯ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1a81:2053 Holtek Semiconductor, Inc. ikbc d4b
Bus 001 Device 003: ID 34d1:0608  USB2.0 HUB
Bus 001 Device 004: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 005: ID 1d57:fa61 Xenta 2.4G Wireless Mouse
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 2b7e:b743 Kingcome HD Webcam
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 003: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 007 Device 006: ID 1a86:8091 QinHeng Electronics USB HUB
Bus 007 Device 007: ID 05e3:0751 Genesys Logic, Inc. microSD Card Reader
Bus 007 Device 008: ID 0c76:153f JMTek, LLC. USB PnP Audio Device
Bus 007 Device 010: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 007 Device 014: ID 1366:1024 SEGGER J-Link_v12
Bus 007 Device 017: ID 1a86:55da QinHeng Electronics USB Dual_Serial
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 002: ID 05e3:0626 Genesys Logic, Inc. Hub
Bus 008 Device 003: ID 0bda:8156 Realtek Semiconductor Corp. USB 10/100/1G/2.5G LAN
Bus 008 Device 004: ID 05e3:0616 Genesys Logic, Inc. hub
```

taotieren, 2024-07-03:
重新断电再进 9008 再刷，听到第一声响后运行刷 uboot 成功

```bash
❯ qsaharaserver -p /dev/ttyUSB0 -s 13:QSahara/mmcblk0p1_0SBL1.bin
Binary build date: Jul  3 2024 @ 14:02:43
QSAHARASERVER CALLED LIKE THIS: 'qsaharaserver -p /dev/ttyUSB0 -s 13:QSahara/mmcblk0p1_0SBL1.bin 'Current working dir: /Actions-OpenWrt/刷机教程/京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双分区刷机+回原厂+USB救砖文件20240510/京东云AX1800-Pro亚瑟_AX6600雅典娜USB救砖
Sahara mappings:
2: amss.mbn
6: apps.mbn
8: dsp1.mbn
10: dbl.mbn
11: osbl.mbn
12: dsp2.mbn
16: efs1.mbn
17: efs2.mbn
20: efs3.mbn
21: sbl1.mbn
22: sbl2.mbn
23: rpm.mbn
25: tz.mbn
28: dsp3.mbn
29: acdb.mbn
30: wdt.mbn
31: mba.mbn
13: QSahara/mmcblk0p1_0SBL1.bin
Requested ID 13, file: "/Actions-OpenWrt/刷机教程/京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双分区刷机+回原厂+USB救砖文件20240510/京东云AX1800-Pro亚瑟_AX6600雅典娜USB救砖/QSahara/mmcblk0p1_0SBL1.bin"
442184 bytes transferred in 0.055428 seconds (7.6081MBps)

File transferred successfully

NOTE: Target requested image 13 which is DeviceProgrammer. Forcing QUIT. This is by design, ** All is well ** SUCCESS!!

Sahara protocol completed
```

taotieren, 2024-07-03:
安装了 win 系统，用 win 试了下，提示刷入成功，看串口日志还是失败没有覆盖 

```bash
***********************************************************************************
* 如果没有安装Qualcomm USB Driver驱动，请先安装                                   *
* USB线连接路由器和电脑，路由器上电会进入Qualcomm Emergency Download mode         *
* 此时设备管理器-端口 (COM 和 LPT) 下会出现Qualcomm HS-USB QDLoader 9008的COM端口 *
* 如果未出现，则需要拆机短接启动电阻，路由器再上电，出现后可松开短接              *
* 输入9008的COM端口号并回车，第一个命令执行后注意观察设备管理器是否已刷新两次     *
* 观察已刷新两次，再等待5秒，如果要进入U-Boot webui此时可以按住reset按键          *
* 然后按任意键执行第二个命令，执行成功后路由器会亮灯                              *
* 也可以不按reset键，U-Boot会直接启动原闪存上的固件                               *
***********************************************************************************
1.京东云AX1800 Pro亚瑟
2.京东云AX6600雅典娜
请输入序号：2
>>>已选择2.京东云AX6600雅典娜

请输入COM端口号：7
>>>输入端口号为COM7
正在上传SBL1。。。
QSaharaServer.exe -p \\.\COM7 -s 13:mmcblk0p1_0SBL1.bin
Binary build date: Jun 25 2019 @ 03:16:15
QSAHARASERVER CALLED LIKE THIS: 'QSaharaServer.ex'Current working dir: E:\git_clone\Actions-OpenWrt\刷机教程\京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双分区刷机+回原厂+USB救砖文件20240510\京东云AX1800-Pro亚瑟_AX6600雅典娜USB救砖\QSahara
Sahara mappings:
2: amss.mbn
6: apps.mbn
8: dsp1.mbn
10: dbl.mbn
11: osbl.mbn
12: dsp2.mbn
16: efs1.mbn
17: efs2.mbn
20: efs3.mbn
21: sbl1.mbn
22: sbl2.mbn
23: rpm.mbn
25: tz.mbn
28: dsp3.mbn
29: acdb.mbn
30: wdt.mbn
31: mba.mbn
13: mmcblk0p1_0SBL1.bin

16:09:45: Requested ID 13, file: "E:\git_clone\Actions-OpenWrt\刷机教程\京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双 分区刷机+回原厂+USB救砖文件20240510\京东云AX1800-Pro亚瑟_AX6600雅典娜USB救砖\QSahara\mmcblk0p1_0SBL1.bin"

16:09:45: 442184 bytes transferred in 0.047000 seconds (8.9723MBps)

16:09:45: File transferred successfully

NOTE: Target requested image 13 which is DeviceProgrammer. Forcing QUIT. This is by design, ** All is well ** SUCCESS!!

16:09:45: Sahara protocol completed

>>>看到最后输出Sahara protocol completed说明SBL1已上传成功
>>>观察到设备管理器刷新两次后，按任意键继续下一步

请按任意键继续. . .
正在上传CDT DEVCFG QSEE RPM APPSBL。。。
QsaharaServer.exe -p \\.\COM7 -s 1:mmcblk0p11_0CDT_1_AX6600_Athena.bin -s 34:mmcblk0p7_0DEVCFG_1_AX6600_Athena.bin -s 25:mmcblk0p5_0QSEE_1_AX6600_Athena.bin -s 23:mmcblk0p9_0RPM_1_AX6600_Athena.bin -s 5:mmcblk0p13_0APPSBL.bin
Binary build date: Jun 25 2019 @ 03:16:15
QSAHARASERVER CALLED LIKE THIS: 'QsaharaServer.ex'Current working dir: E:\git_clone\Actions-OpenWrt\刷机教程\京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双分区刷机+回原厂+USB救砖文件20240510\京东云AX1800-Pro亚瑟_AX6600雅典娜USB救砖\QSahara
Sahara mappings:
2: amss.mbn
6: apps.mbn
8: dsp1.mbn
10: dbl.mbn
11: osbl.mbn
12: dsp2.mbn
16: efs1.mbn
[... 195 lines cut ...]
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.60 (jdcwifi@A06-R08-I139-137-7000191.JD.LOCAL) (gcc version 5.2.0 (OpenWrt GCC 5.2.0 4.3.0.r4211+r49254) ) #1 SMP PREEMPT Fri Jun 14 20:54:21 CST 2024
[    0.000000] CPU: ARMv7 Processor [51af8014] revision 4 (ARMv7), cr=10c0383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: JDCloud Technologies, Inc. IPQ6018/AP-CP03-C3
[    0.000000] Ignoring memory range 0x40000000 - 0x41000000
[    0.000000] Reserved memory: created DMA memory pool at 0x52f00000, size 24 MiB
[    0.000000] Reserved memory: initialized node dma_pool0@52f00000, compatible id shared-dma-pool
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] PERCPU: Embedded 11 pages/cpu @be6bc000 s15488 r8192 d21376 u45056
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 214308
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 root=PARTUUID=dd27939e-c080-66fb-9451-7e30fa8e4c71 gpt rootwait swiotlb=1 coherent_pool=2M
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 846148K/866304K available (6306K kernel code, 389K rwdata, 1916K rodata, 1024K init, 565K bss, 20156K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
```

lgs2007m, 2024-07-04:
> 安装了 win 系统，用 win 试了下，提示刷入成功，看串口日志还是失败没有覆盖
> 
> ```shell
> ***********************************************************************************
> * 如果没有安装Qualcomm USB Driver驱动，请先安装                                   *
> * USB线连接路由器和电脑，路由器上电会进入Qualcomm Emergency Download mode         *
> * 此时设备管理器-端口 (COM 和 LPT) 下会出现Qualcomm HS-USB QDLoader 9008的COM端口 *
> * 如果未出现，则需要拆机短接启动电阻，路由器再上电，出现后可松开短接              *
> * 输入9008的COM端口号并回车，第一个命令执行后注意观察设备管理器是否已刷新两次     *
> * 观察已刷新两次，再等待5秒，如果要进入U-Boot webui此时可以按住reset按键          *
> * 然后按任意键执行第二个命令，执行成功后路由器会亮灯                              *
> * 也可以不按reset键，U-Boot会直接启动原闪存上的固件                               *
> ***********************************************************************************
> 1.京东云AX1800 Pro亚瑟
> 2.京东云AX6600雅典娜
> 请输入序号：2
> >>>已选择2.京东云AX6600雅典娜
> 
> 
> 请输入COM端口号：7
> >>>输入端口号为COM7
> 正在上传SBL1。。。
> QSaharaServer.exe -p \\.\COM7 -s 13:mmcblk0p1_0SBL1.bin
> Binary build date: Jun 25 2019 @ 03:16:15
> QSAHARASERVER CALLED LIKE THIS: 'QSaharaServer.ex'Current working dir: E:\git_clone\Actions-OpenWrt\刷机教程\京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双分区刷机+回原厂+USB救砖文件20240510\京东云AX1800-Pro亚瑟_AX6600雅典娜USB救砖\QSahara
> Sahara mappings:
> 2: amss.mbn
> 6: apps.mbn
> 8: dsp1.mbn
> 10: dbl.mbn
> 11: osbl.mbn
> 12: dsp2.mbn
> 16: efs1.mbn
> 17: efs2.mbn
> 20: efs3.mbn
> 21: sbl1.mbn
> 22: sbl2.mbn
> 23: rpm.mbn
> 25: tz.mbn
> 28: dsp3.mbn
> 29: acdb.mbn
> 30: wdt.mbn
> 31: mba.mbn
> 13: mmcblk0p1_0SBL1.bin
> 
> 16:09:45: Requested ID 13, file: "E:\git_clone\Actions-OpenWrt\刷机教程\京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双 分区刷机+回原厂+USB救砖文件20240510\京东云AX1800-Pro亚瑟_AX6600雅典娜USB救砖\QSahara\mmcblk0p1_0SBL1.bin"
> 
> 16:09:45: 442184 bytes transferred in 0.047000 seconds (8.9723MBps)
> 
> 
> 
> 16:09:45: File transferred successfully
> 
> 
> NOTE: Target requested image 13 which is DeviceProgrammer. Forcing QUIT. This is by design, ** All is well ** SUCCESS!!
> 
> 
> 16:09:45: Sahara protocol completed
> 
> >>>看到最后输出Sahara protocol completed说明SBL1已上传成功
> >>>观察到设备管理器刷新两次后，按任意键继续下一步
> 
> 请按任意键继续. . .
> 正在上传CDT DEVCFG QSEE RPM APPSBL。。。
> QsaharaServer.exe -p \\.\COM7 -s 1:mmcblk0p11_0CDT_1_AX6600_Athena.bin -s 34:mmcblk0p7_0DEVCFG_1_AX6600_Athena.bin -s 25:mmcblk0p5_0QSEE_1_AX6600_Athena.bin -s 23:mmcblk0p9_0RPM_1_AX6600_Athena.bin -s 5:mmcblk0p13_0APPSBL.bin
> Binary build date: Jun 25 2019 @ 03:16:15
> QSAHARASERVER CALLED LIKE THIS: 'QsaharaServer.ex'Current working dir: E:\git_clone\Actions-OpenWrt\刷机教程\京东云AX1800-Pro亚瑟_AX6600雅典娜分区备份+TTL双分区刷机+回原厂+USB救砖文件20240510\京东云AX1800-Pro亚瑟_AX6600雅典娜USB救砖\QSahara
> Sahara mappings:
> 2: amss.mbn
> 6: apps.mbn
[... 216 lines cut ...]
> [    0.000000] Linux version 4.4.60 (jdcwifi@A06-R08-I139-137-7000191.JD.LOCAL) (gcc version 5.2.0 (OpenWrt GCC 5.2.0 4.3.0.r4211+r49254) ) #1 SMP PREEMPT Fri Jun 14 20:54:21 CST 2024
> [    0.000000] CPU: ARMv7 Processor [51af8014] revision 4 (ARMv7), cr=10c0383d
> [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
> [    0.000000] Machine model: JDCloud Technologies, Inc. IPQ6018/AP-CP03-C3
> [    0.000000] Ignoring memory range 0x40000000 - 0x41000000
> [    0.000000] Reserved memory: created DMA memory pool at 0x52f00000, size 24 MiB
> [    0.000000] Reserved memory: initialized node dma_pool0@52f00000, compatible id shared-dma-pool
> [    0.000000] Memory policy: Data cache writealloc
> [    0.000000] psci: probing for conduit method from DT.
> [    0.000000] psci: PSCIv1.0 detected in firmware.
> [    0.000000] psci: Using standard PSCI v0.2 function IDs
> [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
> [    0.000000] PERCPU: Embedded 11 pages/cpu @be6bc000 s15488 r8192 d21376 u45056
> [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 214308
> [    0.000000] Kernel command line: console=ttyMSM0,115200n8 root=PARTUUID=dd27939e-c080-66fb-9451-7e30fa8e4c71 gpt rootwait swiotlb=1 coherent_pool=2M
> [    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
> [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
> [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
> [    0.000000] Memory: 846148K/866304K available (6306K kernel code, 389K rwdata, 1916K rodata, 1024K init, 565K bss, 20156K reserved, 0K cma-reserved, 0K highmem)
> [    0.000000] Virtual kernel memory layout:
> [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
> [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
> ```

9008是从内存启动uboot，不会刷入数据，自己进uboot webui刷你搞坏的分区或者直接刷img镜像。原eMMC上系统分区有数据的，也可以直接启动上面的系统，进系统再恢复你搞坏的分区。看教程，都有说的，不要让我又回答一般教程里面的东西啦

rodren-lion, 2024-07-04:
有刷好 uboot的 emmc镜像吗 或者dd的 .... 分享一下吗... 没有 我自己在刷了 提取一下...

closed 2024-07-06 by lgs2007m