lgs2007m/Actions-OpenWrt #28: 请问大佬的亚瑟uboot支持大雕的开源编译固件么，我用面包多那个uboot是无法使用的

chris7395328, 2024-06-27:
最近开源了所以自己编译了一下，发现是刷不进去，recovery后缀的可以刷进去不过无法保存配置，回去qwrt是正常。就想是不是分区表还是uboot的原因，我尝试刷2g大分区gpt进去发现overlay分区还是变大了的，但是分区数只有23，不知道是哪一步出错了。

lgs2007m, 2024-06-27:
超过100MB的overlay挂载需要f2fs格式支持，可以参考我的config中的eMMC和挂载部分：
CONFIG_PACKAGE_automount=y
CONFIG_PACKAGE_mmc-utils=y
CONFIG_PACKAGE_f2fsck=y
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_mkf2fs=y
CONFIG_PACKAGE_kmod-fs-f2fs=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-mmc=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_gdisk=y
CONFIG_PACKAGE_sgdisk=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_blockdev=y
CONFIG_PACKAGE_losetup=y
CONFIG_PACKAGE_e2fsprogs=y

chris7395328, 2024-06-27:
> 超过100MB的overlay挂载需要f2fs格式支持，可以参考我的config中的eMMC和挂载部分： CONFIG_PACKAGE_automount=y CONFIG_PACKAGE_mmc-utils=y CONFIG_PACKAGE_f2fsck=y CONFIG_PACKAGE_f2fs-tools=y CONFIG_PACKAGE_mkf2fs=y [...]

谢谢解答，就是说我编译把这部分勾选上就可以么，分区表和uboot需不需要都替换成您的？

lgs2007m, 2024-06-27:
> 谢谢解答，就是说我编译把这部分勾选上就可以么，分区表和uboot需不需要都替换成您的？

对，挂载和这些没关系，QWRT你不是正常使用嘛

chris7395328, 2024-06-27:
> 对，挂载和这些没关系，QWRT你不是正常使用嘛

对，qwrt是正常用，不过ssh看分区会提示说备份分区已损毁，主分区正常。
刚刚编译配置看了下，大雕那个Extra packages里面有automount勾选，然后外面只有f2fsck，你列表剩下那些好像没有全部对得上，是仓库区别么

lgs2007m, 2024-06-27:
> 对，qwrt是正常用，不过ssh看分区会提示说备份分区已损毁，主分区正常。 刚刚编译配置看了下，大雕那个Extra packages里面有automount勾选，然后外面只有f2fsck，你列表剩下那些好像没有全部对得上，是仓库区别么

你没有用我的教程去新建分区和保存分区表，所以有这个提示。
仓库勾选哪些我没看，我只按我的使用习惯，把需要的勾选上。我编译的固件挂载没问题，不过USB还有问题，等大佬修复。

chris7395328, 2024-06-27:
> 你没有用我的教程去新建分区和保存分区表，所以有这个提示。 仓库勾选哪些我没看，我只按我的使用习惯，把需要的勾选上。我编译的固件挂载没问题，不过USB还有问题，等大佬修复。

好的谢谢，我确实是漏了再保存分区表这个操作，当时是ttl备份，就直接刷uboot和gpt了，刚刚才看到您的教程说刷了固件进去后要再保存一次分区表

lgs2007m, 2024-06-27:
> > > > > > 超过100MB的overlay挂载需要f2fs格式支持，可以参考我的config中的eMMC和挂载部分： CONFIG_PACKAGE_automount=y CONFIG_PACKAGE_mmc-utils=y CONFIG_PACKAGE_f2fsck=y CONFIG_PACKAGE_f2fs-tools=y CONFIG_PACKAGE_mkf2fs=y CONFIG_PACKAGE_kmod-fs-f2fs=y CONFIG_PACKAGE_kmod-fs-ext4=y CONFIG_PACKAGE_kmod-mmc=y CONFIG_PACKAGE_lsblk=y CONFIG_PACKAGE_fdisk=y CONFIG_PACKAGE_gdisk=y CONFIG_PACKAGE_sgdisk=y CONFIG_PACKAGE_blkid=y CONFIG_PACKAGE_blockdev=y CONFIG_PACKAGE_losetup=y CONFIG_PACKAGE_e2fsprogs=y
> > > > > 
> > > > > 
> > > > > 谢谢解答，就是说我编译把这部分勾选上就可以么，分区表和uboot需不需要都替换成您的？
> > > > 
> > > > 
> > > > 对，挂载和这些没关系，QWRT你不是正常使用嘛
> > > 
> > > 
> > > 对，qwrt是正常用，不过ssh看分区会提示说备份分区已损毁，主分区正常。 刚刚编译配置看了下，大雕那个Extra packages里面有automount勾选，然后外面只有f2fsck，你列表剩下那些好像没有全部对得上，是仓库区别么
> > 
> > 
> > 你没有用我的教程去新建分区和保存分区表，所以有这个提示。 仓库勾选哪些我没看，我只按我的使用习惯，把需要的勾选上。我编译的固件挂载没问题，不过USB还有问题，等大佬修复。
> 
> 好的谢谢，我确实是漏了再保存分区表这个操作，当时是ttl备份，就直接刷uboot和gpt了，刚刚才看到您的教程说刷了固件进去后要再保存一次分区表。可以顺便问问大佬lede仓库编译的固件，recovery固件和factory固件有什么区别么，为啥factory的反而刷不进去了

大雕说以后可能出不支持原厂固件的uboot，kernel也采用更大的，kernel分区也调整为12M的（原厂分区表kernel是6M）。我猜测基于这个原因，他将lede仓库中亚瑟雅典娜的factory固件kernel从原来6M升为了12M，新增一个recovery固件保留kernel为6M，因此现在的uboot刷lede仓库原始配置编译的固件，刷recovery固件

closed 2024-06-27 by lgs2007m

chris7395328, 2024-06-27:
> CONFIG_PACKAGE_sgdisk

感谢，把f2fsck那些编译好后一切正常了