lgs2007m/Actions-OpenWrt #30: JDC-AX1800 Pro亚瑟云编译固件没有WiFi

laipeng668, 2024-06-28:
云编译固件没有WiFi选项，我自己刷京东官方、本地编译、QWRT均有无线选择。
[image]

lgs2007m, 2024-06-28:
默认是有的，我云编译的也有。不过开源的亚瑟、雅典娜USB还有问题，不建议使用。
不是用我的config配置的话，可以看看我的

closed 2024-06-28 by lgs2007m

laipeng668, 2024-06-28:
> 默认是有的，我云编译的也有。不过开源的亚瑟、雅典娜USB还有问题，不建议使用。 不是用我的config配置的话，可以看看我的

确实是有的，系统日志启动了，重置后第一次启动，15分钟后才会出现无线选项，不过不影响使用，谢谢！

> Thu Jun 27 15:57:39 2024 kern.info kernel: [   17.659771] ath11k c000000.wifi: ipq6018 hw1.0
Thu Jun 27 15:57:39 2024 kern.info kernel: [   17.659806] ath11k c000000.wifi: FW memory mode: 1
Thu Jun 27 15:57:39 2024 kern.info kernel: [   17.663539] remoteproc remoteproc0: powering up cd00000.remoteproc
Thu Jun 27 15:57:39 2024 kern.info kernel: [   17.710728] remoteproc remoteproc0: Booting fw image IPQ6018/q6_fw.mdt, size 7340
Thu Jun 27 15:57:39 2024 kern.info kernel: [   18.495388] remoteproc remoteproc0: remote processor cd00000.remoteproc is now up
Thu Jun 27 15:57:39 2024 kern.warn kernel: [   18.496634] ath11k c000000.wifi: qmi ignore invalid mem req type 3
Thu Jun 27 15:57:39 2024 user.info kernel: [   18.498036] kmodloader: done loading kernel modules from /etc/modules.d/*
Thu Jun 27 15:57:39 2024 kern.info kernel: [   18.502404] ath11k c000000.wifi: chip_id 0x0 chip_family 0x4 board_id 0xff soc_id 0xffffffff
Thu Jun 27 15:57:39 2024 kern.info kernel: [   18.514821] ath11k c000000.wifi: fw_version 0x250a04b8 fw_build_timestamp 2021-12-20 07:20 fw_build_id WLAN.HK.2.5.0.1-01208-QCAHKSWPL_SILICONZ-1