lgs2007m/Actions-OpenWrt #31: 请问下大佬，怎么把2.5G网口设置为wan口，然后在把第一个lan口设置为wan口

iibowen, 2024-06-29:
请问下大佬，怎么把2.5G网口设置为wan口，然后在把第一个lan口设置为wan口

lgs2007m, 2024-06-29:
自己谷歌，不做使用指导。

closed 2024-06-29 by lgs2007m

lgs2007m, 2024-08-16:
最近需要用路由器2.5G做测速服务，改了下wan为lan，靠近wan的lan1改为wan，你可以参考看看
[image]

修改后的/etc/config/network文件中的mac是我自己的，你改为你的就行，改完reboot重启路由器
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'
	list ports 'eth1'

config device
	option name 'eth0'
	option macaddr 'dc:d8:7c:55:55:50'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'eth0.2'
	option proto 'dhcp'
	option macaddr 'dc:d8:7c:55:55:51'

config interface 'wan6'
	option device 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '2 3 4 6t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '1 6t'

config interface 'docker'
	option device 'docker0'
	option proto 'none'
	option auto '0'

config device
	option type 'bridge'
	option name 'docker0'

```