lgs2007m/Actions-OpenWrt #34: 无灯又进不了9008

yklee63, 2024-07-03:
请教大佬：
新买的京东云亚瑟128G路由器，因为没有接触过，按照https://phyng.com/2024/05/05/ax1800-openwrt.html的方法刷了uboot和iStoreOS工厂版，扩大了overlay分区。
然后想刷该文介绍的openwrt.sysupgrade.bin(38M)，却失败了，进不了后台界面，也没有wifi。于是改刷Fabric.bin,立刻成红灯变砖。因为是在uboot下刷机，估计不容易烧芯片，于是想进入uboot,按住reset数秒钟后开电，这次直接灯也不亮了。拆机后又试着断电按住reset几秒钟后开电，此时由长红灯转为闪烁红灯五次，然后变蓝灯。应该是进入了uboot，不过用网线连接却进入不了192.168.1.1Uboot界面。但是可以用TTL跑码，中断后：
Hit any key to stop autoboot:  0
IPQ6018# printenv
baudrate=115200
bootargs=console=ttyMSM0,115200n8
bootcmd=bootipq; httpd 192.168.1.1; reset
bootdelay=3
eth1addr=dc:d8:7c:57:74:b1
eth2addr=dc:d8:7d:57:74:b1
eth3addr=dc:d8:7e:57:74:b1
eth4addr=dc:d8:7c:57:74:b2
ethact=eth0
ethaddr=dc:d8:7b:57:74:b1
fdt_high=0x48500000
fdtcontroladdr=4a479c10
flash_type=5
ipaddr=192.168.1.1
machid=8030201
netmask=255.255.255.0
serverip=192.168.1.2
soc_version_major=1
soc_version_minor=0
stderr=serial@78B1000
stdin=serial@78B1000
stdout=serial@78B1000

Environment size: 534/262140 bytes
IPQ6018#
本来可以刷回原系统，因为辗转找到大佬的救砖方法，以及Uboot和固件都比较完美，准备改刷大佬的系统，为安全起见现在网上买一个1.8v的USB转TTL工具，没有进行刷机。等了几天，到了以后接上再试（只有G,R,T三根线），开电却又是无灯，不启动了。TTL也不行了。将主板拆出来，接点太密太小，自己眼睛已经看不清楚，只好在网上找人修理，据说短接了也进入不了9008模式，退了回来。
我想自始至终都是在Uboot下刷机，也没有在TTL下进行刷机，CPU应该不会损坏吧。还有什么可以指教的地方请多多指教。

lgs2007m, 2024-07-03:
你是新机，这就不懂……uboot下刷什么固件都不会无灯，因为上电加载uboot就会有红灯，除非是用我的uboot，进入uboot、img页面刷了其他uboot或者img镜像、分区表。无灯可能前面的分区出了问题，需要9008重新救。9008都不行，那可能电源或者CPU出问题了。当然也不排除网上找的人不会修。这个你自己排查吧。先上电看TTL是否跑马，不跑码就得上USB 9008，如果9008没反应就是硬件坏了，可能电源也可能CPU

yklee63, 2024-07-04:
谢谢指教。如果短接点滑到别的位置有可能烧CPU吗？

lgs2007m, 2024-07-04:
> 谢谢指教。如果短接点滑到别的位置有可能烧CPU吗？

你这问题无法回答，只能说有可能

closed 2024-07-04 by lgs2007m