lgs2007m/Actions-OpenWrt #35: 雅典娜刷uboot后有点问题

magisk317, 2024-07-04:
首先整体操作应该是成功的，uboot刷上了，也可以进去webui刷各个版本的官方固件，但是有几个小问题

1. 空间大小识别错误
点贡献度，最下边的设备信息，显示32G，机器是256的
2. 无法跑分
路由器是今天刚买的，然后就开了ssh刷了uboot，想法很简单，保留ssh，保留uboot，然后可以正常跑分
现在的情况是，只有uboot是正常的，但是随便换个固件ssh就没了，这个问题也还好，固化问题，可能得后续再研究
主要是不能跑分，试过最老的1.5.40.r2181：
[image]
也试过最新的4.3.0.r4211，都不行
首页提示服务异常，内置存储设置为智能加速的时候提示失败，如下图
[image]
尝试各个版本刷机，无果，都一样，刚开始的时候还刷了双分区分区表，以为是这个问题，后来dd刷回机器本身的也不行。。

所做的尝试就上述这些，实在高不定，特来求助。。

magisk317, 2024-07-04:
貌似是[这里](https://github.com/lgs2007m/Actions-OpenWrt/blob/main/%E5%88%B7%E6%9C%BA%E6%95%99%E7%A8%8B/%E4%BA%AC%E4%B8%9C%E4%BA%91AX1800-Pro%E4%BA%9A%E7%91%9F_AX6600%E9%9B%85%E5%85%B8%E5%A8%9C%E5%88%86%E5%8C%BA%E5%A4%87%E4%BB%BD%2BTTL%E5%8F%8C%E5%88%86%E5%8C%BA%E5%88%B7%E6%9C%BA%2B%E5%9B%9E%E5%8E%9F%E5%8E%82%2BUSB%E6%95%91%E7%A0%96%E6%95%99%E7%A8%8B20240510.md#5%E6%96%B0%E5%BB%BAstorage%E5%88%86%E5%8C%BA%E5%B9%B6%E8%BF%98%E5%8E%9F%E8%B7%91%E5%88%86%E5%88%86%E5%8C%BA)的问题，但是我这个装不上，什么鬼？明天研究一下
```
#在线：
root@JDBox_Athena:/# opkg install sgdisk
Unknown package 'sgdisk'.
Collected errors:
 * opkg_install_cmd: Cannot install package sgdisk.

#离线
root@JDBox_Athena:/# opkg install /tmp/sgdisk_1.0.6-1_aarch64_cortex-a53.ipk
Unknown package 'sgdisk'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for sgdisk found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package sgdisk.
```

rodren-lion, 2024-07-04:
跑分 还刷 uboot...   我的雅典娜 跑了2年快退役了..... 二十几万豆不知道 emmc 快挂了没准备开刷了..

magisk317, 2024-07-04:
> 跑分 还刷 uboot... 我的雅典娜 跑了2年快退役了..... 二十几万豆不知道 emmc 快挂了没准备开刷了..

哈哈哈哈哈，留个后路嘛，别跟之前的一样，傻乎乎的升级，到最后想刷都刷不了了，气死

lgs2007m, 2024-07-05:
> 貌似是[这里](https://github.com/lgs2007m/Actions-OpenWrt/blob/main/%E5%88%B7%E6%9C%BA%E6%95%99%E7%A8%8B/%E4%BA%AC%E4%B8%9C%E4%BA%91AX1800-Pro%E4%BA%9A%E7%91%9F_AX6600%E9%9B%85%E5%85%B8%E5%A8%9C%E5%88%86%E5 [...]

需要刷op来安装sgdisk，原厂系统不能装的。
如果跑原厂，不需要刷分区表，只刷uboot即可，而且必选刷最新的4.3.0.r4211，你不知道什么时候会被强制更新。如果要扩容后回原厂，按教程一步一步来，先刷好uboot和分区表，刷op在进系统新建storage分区，恢复其他跑分分区。

这些教程都有说明的，你不按教程来，没什么好说的

closed 2024-07-05 by lgs2007m

magisk317, 2024-07-07:
@lgs2007m  感谢大佬，已经刷了最新版跑上了，再请教一个问题，如果后续再有新版本，是直接在uboot刷即可，还是跟之前一样，需要还原分区，新建storage？（扩容的情况）

lgs2007m, 2024-07-07:
> @lgs2007m 感谢大佬，已经刷了最新版跑上了，再请教一个问题，如果后续再有新版本，是直接在uboot刷即可，还是跟之前一样，需要还原分区，新建storage？（扩容的情况）

只需uboot刷即可。刷分区表是一劳永逸的方法，只需一次。