lgs2007m/Actions-OpenWrt #37: r4211 9008刷机

LiangLin0316, 2024-07-06:
按照作者的教程，短接电阻，短接后灯灭了  串口也显示9008，
运行脚本，上传SBL1 然后继续，此时路由器指示灯会亮起 （拔掉对公USB了）马上按住reset按键，等待红灯闪烁，变为蓝灯可以松开reset，蓝灯时候松开reset键 ，访问192.168.1.1 已经可以正常访问uboot，此时上传固件，上传完路由器绿灯亮  好像重启的时候又起不来了 ，灯是灭的，上面红灯亮的时候我不更新固件 然后拔掉路由器电源再插上 灯也还是灭的 ，路由器起不来了 ，插上对公数据线 还是现实9008串口。 还是可以重复上面流程 进uboot 但是拔掉电源重启 就起不来 ， 请教下作者 这是什么原因啊

lgs2007m, 2024-07-06:
要先用9008刷uboot，再正常进uboot刷固件。按理说说也9008可以刷固件，但是不知道你刷的是什么固件，原厂uboot是否能启动你的固件？

你现在插电连灯都不亮？那就是砖了，按理说9008刷用我改的uboot刷固件不会有问题，不清楚你搞了什么哦。有TTL log吗？

LiangLin0316, 2024-07-06:
先用9008刷uboot,刷完直接按reset按键 然后就可以访问192.168.1.1上传固件  但是上传固件完路由器重启  灯点不亮了  现在只要插上对公usb就默认是9008模式 ，然后还是可以刷UBOOT 只是刷完重启还是灯不亮重启不了 ，灯只会在Usb命令.bat执行完才会亮，才能启动路由器 然后重启就起不来了

lgs2007m, 2024-07-06:
你已经砖了，自己看教程用备份文件合并一个gpt到appsbl_1的镜像刷入吧

LiangLin0316, 2024-07-06:
/bin/ash: ▒: not found
root@iStoreOS:/#
root@iStoreOS:/#
root@iStoreOS:/#
root@iStoreOS:/#
root@iStoreOS:/#
root@iStoreOS:/#
root@iStoreOS:/#
root@iStoreOS:/#
root@iStoreOS:/# reboot
root@iStoreOS:/# [  575.078766] br-lan: port 5(ath1) entered disabled state
[  575.078842] br-lan: port 4(ath0) entered disabled state
[  575.082843] br-lan: port 3(eth2) entered disabled state
[  575.100832] device ath0 left promiscuous mode
[  575.100918] br-lan: port 4(ath0) entered disabled state
[  575.135292] device ath1 left promiscuous mode
[  575.135363] br-lan: port 5(ath1) entered disabled state
[  575.185168] device eth0 left promiscuous mode
[  575.185248] br-lan: port 1(eth0) entered disabled state
[  575.252077] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[  575.252200] device eth1 left promiscuous mode
[  575.256900] br-lan: port 2(eth1) entered disabled state
[  575.322101] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[  575.322231] device eth2 left promiscuous mode
[  575.327123] br-lan: port 3(eth2) entered disabled state
[  575.382686] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  577.426547] wlan: [787:I:ANY] Skip bcast de-auth reason code 3, handled in VD                                                                                      EV down
[  577.751626] wlan: [787:I:ANY] Skip bcast de-auth reason code 3, handled in VD                                                                                      EV down
[  580.441073] reboot: Restarting system

LiangLin0316, 2024-07-06:
ttl看不到  信息 然后用USB命令.bat能启动  192.168.1.1上传固件好像生效了  可以进istoreOs 但是重启不亮灯  。。救命  每次想用路由器得用USB命令.bat引导开机  合成的话mmcblk0_GPT.bin这个文件在哪啊 我怎么拿到 不太懂  能加个联系方式教教我吗

lgs2007m, 2024-07-06:
已经告诉你方法了，看教程自己做吧。没空帮你

closed 2024-07-06 by lgs2007m