lgs2007m/Actions-OpenWrt #41: 请问 RAX3000M uboot 下如何备份 boot0/boot1 分区？

bin456789, 2024-07-16:
我的RAX3000M是20240115批次，无法解密配置文件，应该是移动换了加密密码
通过ttl 救砖方式顺利进入 uboot ，也通过了mmc和tftpput 命令备份了原系统
```
mmc read 0x46000000 0x00002000 0x00000400 && tftpput 0x46000000 0x00080000 1.bin
...
```
但好像不能读取 boot0/boot1分区？
请问 uboot 下如何备份 boot0/boot1 分区？

lgs2007m, 2024-07-16:
没听说换了新的密码，有TTL可以在备份配置文件的时候，查看TTL输出，密码是在TTL输出日志里的。
RAX3000M算力版中eMMC的boot0分区是bl2，不需要备份。TTL救砖直接刷uboot，uboot刷系统，进系统再备份boot0也行。
RAX3000M算力版刷机不需要刷bl2，只需要刷uboot即可。百里才需要bl2、uboot一起刷。

closed 2024-07-16 by lgs2007m

bin456789, 2024-07-16:
```
MT7981> mmc dev 0 1
switch to partitions #1, OK
mmc0(part 1) is current device

MT7981> mmc read 0x46000000 0x0000 0x1fff
MMC read: dev # 0, block # 0, count 8191 ... 8191 blocks read: OK

MT7981> mmc read 0x46000000 0x0000 0x2000
MMC read: dev # 0, block # 0, count 8192 ... 0 blocks read: ERROR
```

不知道为什么这种方法备份不了boot0最后一个扇区
纯属好奇，完整的boot0我之前备份好了

还有我这批次备份配置时ttl是看不到密码的，论坛也有相关案例
https://www.right.com.cn/forum/thread-8378325-1-1.html
https://www.right.com.cn/forum/thread-8382097-1-1.html

lgs2007m, 2024-07-16:
我没有RAX3000M算力版机子了，用百里进行的测试，是正常的：
`MT7986> mmc dev 0 0
switch to partitions #0, OK
mmc0(part 0) is current device
MT7986> mmc read 0x46000000 0x0000 0x2000

MMC read: dev # 0, block # 0, count 8192 ... 8192 blocks read: OK
MT7986> mmc dev 0 1
switch to partitions #1, OK
mmc0(part 1) is current device
MT7986> mmc read 0x46000000 0x0000 0x2000

MMC read: dev # 0, block # 0, count 8192 ... 8192 blocks read: OK
`
感谢提供信息，20240115批次真有部分机子的配置不一样，改了密码，这个得找到密码，不然目前只能TTL刷机了。
你备份了系统分区，可以把系统分区和配置文件上传恩山给大家看看，看看有没有办法找出密码。

bin456789, 2024-07-16:
[a.zip](https://github.com/user-attachments/files/16251145/a.zip)

固件显示是2月6号编译的，比生产时间晚
最屌的是，里面没有 dropbear

我这个是咸鱼上收的二手
可能上家使用时被移动远程更新过固件

bin456789, 2024-07-16:
哦对了，我重新进入救砖uboot，备份 boot0/boot1也没报错了

lgs2007m, 2024-07-17:
看了下备份加密文件，现在密码不是直接写出来了，用了生成密码的函数，我还看不出怎么弄的。等恩山大佬看看吧

bin456789, 2024-10-28:
牛逼