lgs2007m/Actions-OpenWrt #47: 京东云百里是否完整支持kvr协议无缝漫游？

yushu280, 2024-07-29:
谢谢大佬的7.15固件，用起来很舒服，就是有个美中不足之处，无线网络设置中只看到k，没看到v和r。那么，到底支不支持v和r？是否完整支持kvr协议无缝漫游？

lgs2007m, 2024-08-04:
hanwckf大佬闭源支持k v，v默认开启，还不支持r。你可以试试大雕的QWRT

closed 2024-08-04 by lgs2007m

lgs2007m, 2024-08-15:
对了，237大佬的仓库源码添加了r的支持，可以自己编译看看
https://github.com/padavanonly/immortalwrt-mt798x

Narut0123, 2024-08-15:
> 对了，237大佬的仓库源码添加了r的支持，可以自己编译看看 https://github.com/padavanonly/immortalwrt-mt798x

加菲大佬大概啥时更新百里的固件啊？

lgs2007m, 2024-08-15:
> > 对了，237大佬的仓库源码添加了r的支持，可以自己编译看看 https://github.com/padavanonly/immortalwrt-mt798x
> 
> 加菲大佬大概啥时更新百里的固件啊？

不更啦，追新可以用237大佬的源码编译

Narut0123, 2024-08-15:
> > > 对了，237大佬的仓库源码添加了r的支持，可以自己编译看看 https://github.com/padavanonly/immortalwrt-mt798x
> > 
> > 
> > 加菲大佬大概啥时更新百里的固件啊？
> 
> 21.02万年不变了，不更啦，追新可以用237大佬的源码编译，他还有一些功能性更新

啊这！好吧

lgs2007m, 2024-08-29:
新编译的固件应该是支持kvr了，你可以试试，我不懂不测试了
https://github.com/lgs2007m/Actions-OpenWrt/releases/tag/2024.08.29-1258-hanwckf-jdcloud_re-cp-03-gsw-mtwifi-docker