lgs2007m/Actions-OpenWrt #51: 京东云百里AX6000每次升级都不建议保留配置吗？

Narut0123, 2024-09-11:
京东云百里AX6000 从之前的固件（2024.08.29-1258-hanwckf-jdcloud_re-cp-03）更新到最新版（2024.09.10-1901-hanwckf-jdcloud_re-cp-03），可以保留配置吗？什么情况下不保留配置？

lgs2007m, 2024-09-11:
这次9.10用的是hanwckf大佬的仓库源码，无线没有802.11R漫游支持，其他的相同，没必要更新，因为21.02软件库很久没更新了。如果插件基本不变，可以保留配置，这次9.10的插件和8.29一样，但是无线有点差别，我不知道是否可以保留配置。我都是全新刷写，然后用uci命令恢复配置。你可以直接保留配置，有问题再说哈

closed 2024-09-11 by lgs2007m

Narut0123, 2024-09-11:
> 的是hanwckf大佬的仓库源码，无线没有802.11R漫游支持，其他的相

了解了。多谢大佬答疑

Narut0123, 2024-09-11:
> uci命令恢复配置

请问uci命令导出和恢复配置具体该怎么写？加菲猫大佬可否给个示例？谢谢

lgs2007m, 2024-09-11:
> > uci命令恢复配置
> 
> 请问uci命令导出和恢复配置具体该怎么写？加菲猫大佬可否给个示例？谢谢

uci命令问下chatgpt就行了，再试试
我命令是：
```
#-------------------/etc/config/fstab----------------------
##按照我改的单分区表，最后一个分区是mount[4]，设置mount[4]挂载为/opt给docker用
uci set fstab.@mount[4].target='/opt'
uci set fstab.@mount[4].enabled='1'

uci commit fstab
#/etc/init.d/fstab restart

#-------------------/etc/config/network----------------------
##删除内网IPv6前缀分配
uci delete network.globals.ula_prefix
##设置IP网段192.168.6.1
uci set network.lan.ipaddr='192.168.6.1'
uci set network.wan.proto='pppoe'
uci set network.wan.username='账户'
uci set network.wan.password='密码'

uci commit network
#/etc/init.d/network restart

##设置WireGuard
uci set network.WireGuard=interface
uci set network.WireGuard.proto='wireguard'
uci set network.WireGuard.private_key='路由器端私钥'
uci set network.WireGuard.listen_port='端口号'
uci add_list network.WireGuard.addresses='192.168.55.1/24' ##WireGuard网段
uci add network wireguard_WireGuard
uci set network.@wireguard_WireGuard[-1].description='MyPeer' ##对端名称
uci set network.@wireguard_WireGuard[-1].public_key='对端公钥'
uci set network.@wireguard_WireGuard[-1].preshared_key='预共享密钥'
uci add_list network.@wireguard_WireGuard[-1].allowed_ips='192.168.55.6' ##允许IP

uci commit network
#/etc/init.d/network restart

##WireGuard加入防火墙lan zone
uci add_list firewall.@zone[0].network='WireGuard'

uci commit firewall
#/etc/init.d/firewall restart

#-------------------/etc/config/system----------------------
##关闭LED
uci add system led
uci set system.@led[-1].name='green'
uci set system.@led[-1].sysfs='green:status'
uci set system.@led[-1].trigger='none'
uci set system.@led[-1].default='0'
uci commit system

#/etc/init.d/system restart

#-------------------/etc/config/dhcp----------------------
##修改IP分配从5开始分配，分配限制15个
uci set dhcp.lan.start='5'
uci set dhcp.lan.limit='15'
#uci set dhcp.lan.force='1'
#uci set dhcp.lan.ra='server'
#uci set dhcp.lan.dhcpv6='server'
#uci set dhcp.lan.ra_default='1'
#uci delete dhcp.lan.ndp

uci commit dhcp
#/etc/init.d/dnsmasq restart
#/etc/init.d/odhcpd restart

#-------------------/etc/wireless/mediatek/----------------------
##设置2.4G无线
uci set wireless.MT7986_1_1.channel='1'
uci set wireless.default_MT7986_1_1.ssid='2.4GWiFi名称'
uci set wireless.default_MT7986_1_1.encryption='sae-mixed'
uci set wireless.default_MT7986_1_1.key='密码'
uci set wireless.default_MT7986_1_1.mumimo_dl='0'
uci set wireless.default_MT7986_1_1.mumimo_ul='0'
uci set wireless.MT7986_1_2.channel='36'
uci set wireless.default_MT7986_1_2.ssid='5GWiFi名称'
uci set wireless.default_MT7986_1_2.encryption='sae-mixed'
uci set wireless.default_MT7986_1_2.key='密码'
uci set wireless.default_MT7986_1_2.mumimo_dl='0'
uci set wireless.default_MT7986_1_2.mumimo_ul='0'

uci commit wireless
#/etc/init.d/network restart

#-------------------/etc/config/ddns------------------------------
##设置WireGuard的IPv6 DDNS
uci add ddns service
uci rename ddns.@service[-1]='WG_IPv6'
uci set ddns.@service[-1].enabled='1'
uci set ddns.@service[-1].lookup_host='域名'
uci set ddns.@service[-1].use_ipv6='1'
uci set ddns.@service[-1].service_name='DDNS提供商'
uci set ddns.@service[-1].domain='域名'
uci set ddns.@service[-1].username='账户'
uci set ddns.@service[-1].password='密码'
uci set ddns.@service[-1].use_https='1'
uci set ddns.@service[-1].cacert='/etc/ssl/certs'
uci set ddns.@service[-1].ip_source='interface'
uci set ddns.@service[-1].ip_interface='pppoe-wan' ##IPv6来源拨号的WAN
uci set ddns.@service[-1].interface='pppoe-wan'
uci set ddns.@service[-1].dns_server='223.5.5.5'

uci commit ddns
#/etc/init.d/ddns restart

#-------------------/etc/config/firewall----------------------
##开启WireGuard的IPv6端口
uci add firewall rule
uci set firewall.@rule[-1].name='WG_IPv6'
uci set firewall.@rule[-1].family='ipv6'
uci set firewall.@rule[-1].proto='udp'
uci set firewall.@rule[-1].src='wan'
uci set firewall.@rule[-1].dest_port='端口'
uci set firewall.@rule[-1].target='ACCEPT'

uci commit firewall
#/etc/init.d/firewall restart

##开启upnp
uci set upnpd.config.enabled='1'
uci set upnpd.config.secure_mode='0'
uci commit upnpd

##开启turboacc
uci set turboacc.global.set='1'
uci set turboacc.config.fullcone='1'
uci commit turboacc

##设置WAN口IP，以便方便访问上级桥接的光猫，例如我的光猫网段是192.168.10.1
echo 'ifconfig eth1 192.168.10.2 netmask 255.255.255.0 broadcast 192.168.10.255' >> /etc/firewall.user

##设置路由器密码
echo -e '密码\密码' | passwd root
```
一次性输入，回车执行，然后输入reboot重启即可
其他插件的配置可能复杂点，得慢慢看慢慢修改

Narut0123, 2024-09-11:
> > > uci命令恢复配置
> > 
> > 
> > 请问uci命令导出和恢复配置具体该怎么写？加菲猫大佬可否给个示例？谢谢
> 
> uci命令问下chatgpt就行了，再试试 我命令是：
> 
> ```
> #-------------------/etc/config/fstab----------------------
> ##按照我改的单分区表，最后一个分区是mount[4]，设置mount[4]挂载为/opt给docker用
> uci set fstab.@mount[4].target='/opt'
> uci set fstab.@mount[4].enabled='1'
> 
> uci commit fstab
> #/etc/init.d/fstab restart
> 
> #-------------------/etc/config/network----------------------
> ##删除内网IPv6前缀分配
> uci delete network.globals.ula_prefix
> ##设置IP网段192.168.6.1
> uci set network.lan.ipaddr='192.168.6.1'
> uci set network.wan.proto='pppoe'
> uci set network.wan.username='账户'
> uci set network.wan.password='密码'
> 
> uci commit network
> #/etc/init.d/network restart
> 
> ##设置WireGuard
> uci set network.WireGuard=interface
> uci set network.WireGuard.proto='wireguard'
> uci set network.WireGuard.private_key='路由器端私钥'
> uci set network.WireGuard.listen_port='端口号'
> uci add_list network.WireGuard.addresses='192.168.55.1/24' ##WireGuard网段
> uci add network wireguard_WireGuard
> uci set network.@wireguard_WireGuard[-1].description='MyPeer' ##对端名称
> uci set network.@wireguard_WireGuard[-1].public_key='对端公钥'
> uci set network.@wireguard_WireGuard[-1].preshared_key='预共享密钥'
> uci add_list network.@wireguard_WireGuard[-1].allowed_ips='192.168.55.6' ##允许IP
> 
> uci commit network
> #/etc/init.d/network restart
> 
> ##WireGuard加入防火墙lan zone
> uci add_list firewall.@zone[0].network='WireGuard'
> 
> uci commit firewall
> #/etc/init.d/firewall restart
> 
> #-------------------/etc/config/system----------------------
> ##关闭LED
> uci add system led
> uci set system.@led[-1].name='green'
> uci set system.@led[-1].sysfs='green:status'
> uci set system.@led[-1].trigger='none'
> uci set system.@led[-1].default='0'
> uci commit system
> 
> #/etc/init.d/system restart
> 
> 
> #-------------------/etc/config/dhcp----------------------
> ##修改IP分配从5开始分配，分配限制15个
> uci set dhcp.lan.start='5'
> uci set dhcp.lan.limit='15'
> #uci set dhcp.lan.force='1'
> #uci set dhcp.lan.ra='server'
> #uci set dhcp.lan.dhcpv6='server'
> #uci set dhcp.lan.ra_default='1'
> #uci delete dhcp.lan.ndp
> 
> uci commit dhcp
> #/etc/init.d/dnsmasq restart
> #/etc/init.d/odhcpd restart
> 
> #-------------------/etc/wireless/mediatek/----------------------
> ##设置无线
> uci set wireless.MT7986_1_1.channel='1'
> uci set wireless.default_MT7986_1_1.ssid='2.4GWiFi名称'
> uci set wireless.default_MT7986_1_1.encryption='sae-mixed'
> uci set wireless.default_MT7986_1_1.key='密码'
> uci set wireless.default_MT7986_1_1.mumimo_dl='0'
> uci set wireless.default_MT7986_1_1.mumimo_ul='0'
> uci set wireless.MT7986_1_2.channel='36'
> uci set wireless.default_MT7986_1_2.ssid='5GWiFi名称'
> uci set wireless.default_MT7986_1_2.encryption='sae-mixed'
> uci set wireless.default_MT7986_1_2.key='密码'
> uci set wireless.default_MT7986_1_2.mumimo_dl='0'
> uci set wireless.default_MT7986_1_2.mumimo_ul='0'
> 
> 
> uci commit wireless
> #/etc/init.d/network restart
> 
> #-------------------/etc/config/ddns------------------------------
> ##设置WireGuard的IPv6 DDNS
> uci add ddns service
> uci rename ddns.@service[-1]='WG_IPv6'
> uci set ddns.@service[-1].enabled='1'
> uci set ddns.@service[-1].lookup_host='域名'
> uci set ddns.@service[-1].use_ipv6='1'
> uci set ddns.@service[-1].service_name='DDNS提供商'
> uci set ddns.@service[-1].domain='域名'
> uci set ddns.@service[-1].username='账户'
> uci set ddns.@service[-1].password='密码'
> uci set ddns.@service[-1].use_https='1'
> uci set ddns.@service[-1].cacert='/etc/ssl/certs'
> uci set ddns.@service[-1].ip_source='interface'
> uci set ddns.@service[-1].ip_interface='pppoe-wan' ##IPv6来源拨号的WAN
> uci set ddns.@service[-1].interface='pppoe-wan'
> uci set ddns.@service[-1].dns_server='223.5.5.5'
> 
> uci commit ddns
> #/etc/init.d/ddns restart
> 
> #-------------------/etc/config/firewall----------------------
> ##开启WireGuard的IPv6端口
> uci add firewall rule
> uci set firewall.@rule[-1].name='WG_IPv6'
> uci set firewall.@rule[-1].family='ipv6'
> uci set firewall.@rule[-1].proto='udp'
> uci set firewall.@rule[-1].src='wan'
> uci set firewall.@rule[-1].dest_port='端口'
> uci set firewall.@rule[-1].target='ACCEPT'
> 
> uci commit firewall
> #/etc/init.d/firewall restart
> 
> ##开启upnp
> uci set upnpd.config.enabled='1'
> uci set upnpd.config.secure_mode='0'
> uci commit upnpd
> 
> ##开启turboacc
> uci set turboacc.global.set='1'
> uci set turboacc.config.fullcone='1'
> uci commit turboacc
> 
> ##设置WAN口IP，以便方便访问上级桥接的光猫，例如我的光猫网段是192.168.10.1
> echo 'ifconfig eth1 192.168.10.2 netmask 255.255.255.0 broadcast 192.168.10.255' >> /etc/firewall.user
> 
> ##设置路由器密码
> echo -e '密码\密码' | passwd root
> ```
> 
> 一次性输入，回车执行，然后输入reboot重启即可 其他插件的配置可能复杂点，得慢慢看慢慢修改

万分感谢！Thank you 阿里嘎多 噶撒哈你达！