lgs2007m/Actions-OpenWrt #52: 大佬，什么时候更新23.05的版本

wsjtime, 2024-09-12:
大佬，什么时候更新23.05的版本，现在的版本无法安装Homeproxy。

lgs2007m, 2024-09-12:
可以按主线的百里刷机教程重刷一遍，就可以用主线的固件了，无线开源应该也大差不差。
我目前只用21.02闭源的固件，我也没能力更新版本，等hanwckf大佬更新了

wsjtime, 2024-09-12:
好的，多谢大佬

lgs2007m, 2024-09-16:
> 好的，多谢大佬

更新了gpt和uboot，现在同时支持21.02闭源固件(.bin)和immortalwrt主线master/23.05分支固件(.itb)
https://github.com/lgs2007m/Actions-OpenWrt/releases/tag/2024.09.16-2149-immortalwrt-mainline-support

wsjtime, 2024-09-19:
大佬，re-cp-03，刷了新的UBOOT，GPT都刷完，但是进入UBOOT页面还是以前的UBOOT，刷不了23.5的系统，怎么回事

lgs2007m, 2024-09-19:
> 大佬，re-cp-03，刷了新的UBOOT，GPT都刷完，但是进入UBOOT页面还是以前的UBOOT，刷不了23.5的系统，怎么回事

uboot没刷对吗？核对md5了吗

wsjtime, 2024-09-19:
刷对了，MD5值不一样，我是用的1024的GPT分区表，以前的系统，没有报错

wsjtime, 2024-09-19:
也能进去系统，还是以前的系统

wsjtime, 2024-09-19:
9.18号的UBOOT

wsjtime, 2024-09-19:
刷的这个

wsjtime, 2024-09-19:
按照教程一步步来的

wsjtime, 2024-09-19:
刷的

lgs2007m, 2024-09-19:
我说的是uboot分区的md5和我提供的md5一样吗？

wsjtime, 2024-09-19:
## b1fbbb4b53390f767c62a4088ed3aff5  /dev/mmcblk0boot0
## 556551c8360a71bd05223c021545a20a  /dev/mmcblk0p3
这两个前面的值都不一样，后面的/dev/mmcblk0boot0     /dev/mmcblk0p3 一样

wsjtime, 2024-09-19:
不对，大佬，你新的教程里，BL.2   UBOOT的MD5值好像没写

wsjtime, 2024-09-19:
## 安装教程重新刷一遍，kernel、rootfs及后面的分区数据可能丢失，需要请先做好备份！！！
##
## 【【【重刷bl2和uboot】】】
## 上传bl2和uboot到/tmp文件夹，然后使用下面命令刷写
ash
echo 0 > /sys/block/mmcblk0boot0/force_ro
dd if=/tmp/mt7986_jdcloud_re-cp-03-bl2_mod-fit_20240918.bin of=/dev/mmcblk0boot0 conv=fsync
echo 1 > /sys/block/mmcblk0boot0/force_ro
dd if=/tmp/mt7986_jdcloud_re-cp-03-fip_mod-fit_20240918.bin of=$(blkid -t PARTLABEL=fip -o device) conv=fsync

## 查看md5
md5sum /dev/mmcblk0boot0 && md5sum $(blkid -t PARTLABEL=fip -o device)

## 我的md5，如果你原先不是我的分区表，可能不同，不要紧，命令执行没有报错即可
## root@ImmortalWrt:~# md5sum /dev/mmcblk0boot0 && md5sum $(blkid -t PARTLABEL=fip -o device)
## b1fbbb4b53390f767c62a4088ed3aff5  /dev/mmcblk0boot0
## 556551c8360a71bd05223c021545a20a  /dev/mmcblk0p3

lgs2007m, 2024-09-19:
看uboot页面的日期来判断新旧

lgs2007m, 2024-09-19:
> 大佬，可以加微信吗，BL2和UBOOT输入刷写命令后没有结果显示

点击链接加入QQ群聊【完美闭环-MT798X交流总群】：https://qm.qq.com/q/lyBhaSt8aW

wsjtime, 2024-09-19:
加了，大佬