lgs2007m/Actions-OpenWrt #55: 大佬，9.15的固件，用原来的Uboot刷机失败，怎么回事

wsjtime, 2024-09-15:


lgs2007m, 2024-09-15:
这个是Immortalwrt主线固件6.6内核，测试固件，还有点问题，旧版uboot不支持，等新版uboot吧

closed 2024-09-15 by lgs2007m

wsjtime, 2024-09-15:
好的

lgs2007m, 2024-09-16:
更新了gpt和uboot，现在同时支持21.02闭源固件(.bin)和immortalwrt主线master/23.05分支固件(.itb)
https://github.com/lgs2007m/Actions-OpenWrt/releases/tag/2024.09.16-2149-immortalwrt-mainline-support

wsjtime, 2024-09-17:
好的，大佬，23.05.3和3122c，这两个固件有什么区别？

lgs2007m, 2024-09-17:
> 好的，大佬，23.05.3和3122c，这两个固件有什么区别？
uboot更新了，需要可以看看

原厂固件，hanwckf、237、lean大佬的QWRT都是21.02闭源驱动固件(.bin)，内核是5.4，软件源比较旧了，不支持firewall4
immortalwrt主线master/23.05分支是开源驱动的固件(.itb)，master是6.6内核，23.05是5.15内核，软件源比较新，支持firewall4，支持homeproxy插件

wsjtime, 2024-09-17:
好的，大佬，感谢

fkdndjdn, 2024-09-20:
> > 好的，大佬，23.05.3和3122c，这两个固件有什么区别？
> 
> 今天uboot更新下了，解决了一个问题，需要可以看看下最新的
> 
> 原厂固件，hanwckf、237、lean大佬的QWRT都是21.02闭源驱动固件(.bin)，内核是5.4，软件源比较旧了，不支持firewall4 immortalwrt主线master/23.05分支是开源驱动的固件(.itb)，master是6.6内核，23.05是5.15内核，软件源比较新，支持firewall4，支持homeproxy插件。不过docker不能bridge网络，只能使用host网络，我还不知道什么原因
有没有可能是防火墙的问题。防火墙打开显示iptable和firewall4同时存在，我看iptable规则都是docker的
可能因为这个原因用clash无法正常使用tproxy模式

lgs2007m, 2024-09-20:
> 好的，大佬，23.05.3和3122c，这两个固件有什么区别？
>
>
> 今天uboot更新下了，解决了一个问题，需要可以看看下最新的
> 原厂固件，hanwckf、237、lean大佬的QWRT都是21.02闭源驱动固件(.bin)，内核是5.4，软件源比较旧了，不支持firewall4 [...]

嗯嗯，所以应该怎么解决呢？我不也不懂