lgs2007m/Actions-OpenWrt #59: 刷不进去uboot，正常开机黑灯，无法进入系统

yangmeishu, 2024-10-01:
老大你好，我机器ax1800pro，被我在ssh里调整分区后。重启就黑灯，进不去系统 直接9008。
我在9008里，可以刷gpt，刷固件，但就是刷不进uboot。
现在能通过9008的临时环境来引导进系统，里面能ssh操作等，有什么思路来排查解决这个问题呢

lgs2007m, 2024-10-01:
从描述看可能eMMC有问题了吧，不确定，需要TTL看问题，启动停在哪里

yangmeishu, 2024-10-01:
无法了，按照了你的指示 用ttl查看 可是。看着都正常。这个emmc，我只做分区调整为啥会搞坏呢。再ssh里，如何验证各个分区等是否正常的

`
[2024-10-02 00:49:10.706] Request for: /
[2024-10-02 00:49:10.706] 
[2024-10-02 00:49:10.706] 
[2024-10-02 00:49:10.902] Request for: /style.css
[2024-10-02 00:49:10.902] 
[2024-10-02 00:49:10.902] 
[2024-10-02 00:49:15.846] Request for: /uboot.html
[2024-10-02 00:49:15.846] 
[2024-10-02 00:49:15.846] 
[2024-10-02 00:49:16.042] Request for: /style.css
[2024-10-02 00:49:16.042] 
[2024-10-02 00:49:16.042] 
[2024-10-02 00:49:32.180] Data will be downloaded at 0x50000000 in RAM
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180] Upgrade type: U-Boot
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180] Upload file size: 644624 bytes
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180] Loading: #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #######################################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180]          #################
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180] done!
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.180] 
[2024-10-02 00:49:32.275] Bytes transferred = 644624 (9d610 hex)
[2024-10-02 00:49:32.275] 
[2024-10-02 00:49:32.275] 
[2024-10-02 00:49:32.310] md5 for 50000000 ... 5009d610 ==> 2b37c2b85b009a93a1a01ed80e5bc47b
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] ****************************
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] *     U-BOOT UPGRADING     *
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] * DO NOT POWER OFF DEVICE! *
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] ****************************
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] Executing: mw 0x5009d610 0x00 0x200 && mmc dev 0 && flash 0:APPSBL 0x50000000 $filesize && flash 0:APPSBL_1 0x50000000 $filesize
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.310] 
[2024-10-02 00:49:32.420] switch to partitions #0, OK
[2024-10-02 00:49:32.420] 
[2024-10-02 00:49:32.420] mmc0(part 0) is current device
[2024-10-02 00:49:32.420] 
[2024-10-02 00:49:32.420] 
[2024-10-02 00:49:32.420] 
[2024-10-02 00:49:32.420] MMC erase: dev # 0, block # 13346, count 1280 ... 
[2024-10-02 00:49:32.464] 1280 blocks erased: OK
[2024-10-02 00:49:32.464] 
[2024-10-02 00:49:32.464] 
[2024-10-02 00:49:32.464] 
[2024-10-02 00:49:32.464] MMC write: dev # 0, block # 13346, count 1260 ... 
[2024-10-02 00:49:32.510] 1260 blocks written: OK
[2024-10-02 00:49:32.510] 
[2024-10-02 00:49:32.510] 
[2024-10-02 00:49:32.510] 
[2024-10-02 00:49:32.510] MMC erase: dev # 0, block # 14626, count 1280 ... 1280 blocks erased: OK
[2024-10-02 00:49:32.510] 
[2024-10-02 00:49:32.510] 
[2024-10-02 00:49:32.510] 
[2024-10-02 00:49:32.510] MMC write: dev # 0, block # 14626, count 1260 ... 
[2024-10-02 00:49:32.527] 1260 blocks written: OK
[2024-10-02 00:49:32.527] 
[2024-10-02 00:49:32.527] HTTP ugrade is done! Rebooting...
[2024-10-02 00:49:32.527] 
[2024-10-02 00:49:32.527] 
[2024-10-02 00:49:32.527] 
[2024-10-02 00:49:32.527] 
[2024-10-02 00:49:35.543] resetting ...`

lgs2007m, 2024-10-02:
检查9008的两个焊盘是否被挤压粘连在一起了，需要断电，用小刀轻轻分开。
很多人都出现这个问题，我用镊子短接，一点问题都没有

yangmeishu, 2024-10-02:
也看了，还用万用表通联量了。没通。靠边的那个点和和那个大铜圈有读数是960上下。和另一个点量是通联的，算了。挂着卖或者就拿9008引导算了。哎 
[image]
[image]

lgs2007m, 2024-10-02:
看着焊盘没问题。9008进uboot打开 192.168.1.1/img.html 重刷gpt看看，或者直接刷合并gpt到APPSBL分区的img镜像，最好是你自己的备份。下面这个是我的镜像，刷了之后环境变量的分区APPSBLENV会被我的覆盖，不知道对跑分是否有影响，自行斟酌[JDCloud_AX1800_Pro_IMG.zip](https://github.com/user-attachments/files/17219355/JDCloud_AX1800_Pro_IMG.zip)

yangmeishu, 2024-10-02:
天，老大太强了。我再9008临时web里的img里。上传了你的bin，已经正常。简直是路由器的再生父母。 感谢感谢，溢于言表。林伟老大你用哪个uboot，哪个分区，哪个系统呢，我跟着你一起养老用了。不折腾了

lgs2007m, 2024-10-02:
看教程按自己喜欢的刷，我已经不用亚瑟了

closed 2024-10-02 by lgs2007m