lgs2007m/Actions-OpenWrt #61: 刷第三方OP出错

p0we7, 2024-10-07:
用的是本 repo 的 uboot 和 Actions 编译的固件。

按照教程刷了 bl2和uboot md5都没有问题，然后进 uboot 刷了 actions 编译的 [hanwckf](https://cmi.hanwckf.top/)固件，能成功刷入并进系统。

之后再进 uboot 刷 ImmortalWrt 9.25 actions 编译的固件就出现下面的错误，并且试过刷备份的后的 gpt 固件再刷 9.25固件也不行。

也试过刷 ImmortalWrt Firmware Selector 的 15MB 固件也不行。

请问应该怎么解决？

[image]

lgs2007m, 2024-10-07:
哪里来的9.25？我没有编译过
ImmortalWrt Firmware Selector固件？是.bin格式固件吗？据我了解ImmortalWrt主线已全部切换为.itb格式的FIT image。只有2024.09.18后的uboot配合新的gpt分区表才可以刷主线.itb格式的固件

p0we7, 2024-10-09:
> 哪里来的9.25？我没有编译过 ImmortalWrt Firmware Selector固件？是.bin格式固件吗？据我了解ImmortalWrt主线已全部切换为.itb格式的FIT image。只有2024.09.18后的uboot配合新的gpt分区表才可以刷主线.itb格式的固件

我看错了，是9月15日，本repo编译的固件。 后来根据恩山的方法，刷了 bl2 uboot gpt 分区之后可以刷入了。

lgs2007m, 2024-10-09:
> > 哪里来的9.25？我没有编译过 ImmortalWrt Firmware Selector固件？是.bin格式固件吗？据我了解ImmortalWrt主线已全部切换为.itb格式的FIT image。只有2024.09.18后的uboot配合新的gpt分区表才可以刷主线.itb格式的固件
> 
> 我看错了，是9月15日，本repo编译的固件。 后来根据恩山的方法，刷了 bl2 uboot gpt 分区之后可以刷入了。

建议看我的教程，恕我直言恩山的百里教程和我的比不了
还有不是按我教程刷的，不要来咨询我了……我不帮人排错