lgs2007m/Actions-OpenWrt #63: 今天上午发布的京东云百里固件为何下架了？

Narut0123, 2024-10-15:
今天上午Releases页面看到新发布了一版固件 2024.10.15-0213-hanwckf-jdcloud_re-cp-03-v7672，为何下午下架了？是存在什么问题吗？

lgs2007m, 2024-10-15:
没有问题，可以用，不过无线连接速率显示可能有问题，大佬更新了[mt_wifi: fix 7.6.7.2 rxrate stainfo patch](https://github.com/hanwckf/immortalwrt-mt798x/commit/b869f114102b62005c3868648e1dae5edd3e7694)，重新编译下

closed 2024-10-15 by lgs2007m

Narut0123, 2024-10-15:
> 没有问题，可以用，不过无线连接速率显示可能有问题，大佬更新了[mt_wifi: fix 7.6.7.2 rxrate stainfo patch](https://github.com/hanwckf/immortalwrt-mt798x/commit/b869f114102b62005c3868648e1dae5edd3e7694)，重新编译下
了解了，等待大佬重新编译的新版

Narut0123, 2024-10-16:
大佬您也太高产了吧！昨晚刚刷了2024.10.15-1628版本的百里固件，怎么今早又更新了？😂

lgs2007m, 2024-10-16:
> 大佬您也太高产了吧！昨晚刚刷了2024.10.15-1628版本的百里固件，怎么今早又更新了？😂

hanwckf大佬仓库内核更新到5.4.284了，固件也更新一波咯，现在内核和无线驱动都是新的

Narut0123, 2024-10-16:
> > 大佬您也太高产了吧！昨晚刚刷了2024.10.15-1628版本的百里固件，怎么今早又更新了？😂
> 
> hanwckf大佬仓库内核更新到5.4.284了，固件也更新一波咯，现在内核和无线驱动都是新的。 有更新才会编译，没更新不会重新编译的。

怎么又双叒更新了？😂

Narut0123, 2024-10-16:
> > > > 大佬您也太高产了吧！昨晚刚刷了2024.10.15-1628版本的百里固件，怎么今早又更新了？😂
> > > 
> > > 
> > > hanwckf大佬仓库内核更新到5.4.284了，固件也更新一波咯，现在内核和无线驱动都是新的。 有更新才会编译，没更新不会重新编译的。
> > 
> > 
> > 怎么又双叒更新了？😂
> 
> 测试用的，不要问了…能下载就是能用。不要每次都问！我都懒得回答你

好的！Sorry~~~