lgs2007m/Actions-OpenWrt #64: 京东百里AX6000更新2024.10.16-0916版固件后出现已安装软件消失且更新软件包列表出错

fya16838, 2024-10-17:
作者你好，我在下载最新构建的2024.10.16-0916-hanwckf-jdcloud_re-cp-03-v7672-gsw-docker固件后出现原本安装的软件消失，且无法更新软件包列表，这是什么原因呢？用的系统的更新功能，不是U-BOOT更新的。是不是每次更新都无法保留已安装软件和配置。如果可以的话该如何操作呢？能否解答一下，万分感谢！

[image]
[image]

lgs2007m, 2024-10-17:
不是操作问题，系统中更新固件不会保留之前安装的插件，最多保留之前的配置，这是基础知识。
我使用的是相同固件，更新软件包列表没有问题，检查自己网络。

closed 2024-10-17 by lgs2007m

fya16838, 2024-10-18:
感谢解答，因为我需要一些插件，我在想可不可以自己构建，所以我Fork了你的代码，请问下增加新的软件包是不是如下图位置，格式为：CONFIG_PACKAGE_软件包名称=y
[image]

lgs2007m, 2024-10-18:
是的，添加的插件需要[feeds.conf.default](https://github.com/lgs2007m/Actions-OpenWrt/blob/main/JDCloud-AX6000-Baili-hanwckf/feeds.conf.default)软件源当中的，如果软件源没有，需要修改[diy-part1.sh](https://github.com/lgs2007m/Actions-OpenWrt/blob/main/JDCloud-AX6000-Baili-hanwckf/diy-part1.sh)添加源码，再添加CONFIG_PACKAGE

lgs2007m, 2024-10-18:
自己搜索学习，不做教学

fya16838, 2024-10-18:
好的