lgs2007m/Actions-OpenWrt #93: 百里没有信号

yjiawqgj, 2025-04-19:
按照你的教程刷了，也用了你的固件，可手机就是搜索不到无线信号，换了手机也不行。大佬怎么办？百里64G,RE-CS-05版本。

lgs2007m, 2025-04-19:
怀疑信道设置有问题，设置2.4G信道1，5G信道36

closed 2025-04-19 by lgs2007m

yjiawqgj, 2025-04-19:
依然没搜索不到，而且只能识别两个端口，eth1为wan口，eth0为lan口，也看不到其它三个口，也就无法添加。在概览里能看到五个接口，但是只显示wan口和lan已显示连接，其它三个口没有任何信息。

lgs2007m, 2025-04-19:
> 依然没搜索不到，而且只能识别两个端口，eth1为wan口，eth0为lan口，也看不到其它三个口，也就无法添加。在概览里能看到五个接口，但是只显示wan口和lan已显示连接，其它三个口没有任何信息。

你的factory有没有动过？可以尝试恢复下，关于网口不识别，不要看概览，直接看能不能用。我自用没有遇到过这些问题，你自己排查吧

yjiawqgj, 2025-04-19:
我将保存的原厂的分区的mmcblk0p3_factory.bin文件放到/tmp并执行了dd if=/tmp/mmcblk0p3_factory.bin of=$(blkid -t PARTLABEL=factory -o device) conv=fsync  输出 
4096+0 records in
4096+0 records out
依然没有信号

lgs2007m, 2025-04-19:
> 我将保存的原厂的分区的mmcblk0p3_factory.bin文件放到/tmp并执行了dd if=/tmp/mmcblk0p3_factory.bin of=$(blkid -t PARTLABEL=factory -o device) conv=fsync 输出 4096+0 records in 4096+0 records out 依然没有信号

固件没问题，factory没有问题，那我也不懂了，自己排查吧

yjiawqgj, 2025-04-19:
已解决